shorepine/tulipcc #2: CircuitPython?

tannewt, 2023-02-11:
Any interest in porting this to CircuitPython? I added the broadcom port for exactly this style of use on the pi 400. Biggest hurdle now is usb host over pcie.

dpwe, 2023-02-11:
As I understand it, a big part of the tulip customization of micropython is the graphics -- i.e. interfacing to the framebuffer. That seems like a substantial barrier to porting it to, e.g., RPi hardware.

bwhitman, 2023-02-11:
Yes, welcome @tannewt and thanks @dpwe ! I don't know much about the differences between CircuitPython and Micropython. It may be simple to bolt the bits of Tulip that make it "Tulip" onto CircuitPython. But the bigger challenge will be to get the Tulip display code running on whatever video hardware is connected to the CircuitPython port. The Tulip display code is very much built ontop of a specific ESP chip with GDMA, with a specific type of TTL RGB display. One would have to abstract that out to the CircuitPython frame buffer code. 

I know it's not the same, but I bet Tulip Desktop would compile and run fine for the Raspberry Pi running Linux.

tannewt, 2023-02-13:
> Yes, welcome @tannewt and thanks @dpwe ! I don't know much about the differences between CircuitPython and Micropython.

We have a [quick run down of differences in our README](https://github.com/adafruit/circuitpython/tree/main#differences-from-micropython) if you are curious.
 
> It may be simple to bolt the bits of Tulip that make it "Tulip" onto CircuitPython. But the bigger challenge will be to get the Tulip display code running on whatever video hardware is connected to [...]

Ya, that's true. The broadcom chips are even a bit weirder where the GPU starts up first and then starts the arm cortex A chips. In the end, we do just [get a framebuffer to write to from the gpu though](https://github.com/adafruit/circuitpython/blob/main/ports/broadcom/common-hal/videocore/Framebuffer.c). HDMI support makes so much sense to me since that'll make this a "plug into your tv or monitor" computer like the C64 era.

For ESP we don't support 2d dma or TTL rgb yet. However, we have pretty universal display API. We're looking into having a separate RP2040 convert from SPI to DVI over HDMI. The higher end iMX RT chips also have 2D dma iirc.

> 
> I know it's not the same, but I bet Tulip Desktop would compile and run fine for the Raspberry Pi running Linux.

Definitely not the same. But point taken.

dglaude, 2023-02-19:
Just a message to say that I love the Tulip Computer project (that I discovered thanks to Adafruit Python on Hardware newsletter).

The concept of a computer booting into (Circuit) Python has been in my head since a long time and I described it here early 2021: https://learn.adafruit.com/u/dglaude/pages/circuitpython2021-the-year-of-the-circuitpython-retro-gaming-personal-computer

I don't think I am going to build an hardware copy, maybe just because of the costly screen you are using. Right now I prefer the idea of a processor dedicated to video displaying, especially Pico/RP2040 with VGA or DVI as described in my 2023 version: https://learn.adafruit.com/u/dglaude/pages/circuitpython2023-the-year-of-the-circuitpython-retro-gaming-personal-computer

The co-processor for screen display idea is also used in "Agon light"(TM) computer https://www.thebyteattic.com/p/agon.html (similar and different, it is build around eZ80 8bit computer, run BBC Basic, but handle the video output = VGA with a dedicated ESP32-PICO-D4.

bwhitman, 2023-02-20:
Very cool. I'm going to move this over from issues to discussions so we can keep issues about fixes / problems with Tulip! See you over there