shorepine/tulipcc #14: macOS build fails with "modffi.c:32:10: fatal error: 'ffi.h' file not found"

blackketter, 2023-02-23:
Trying to build tulipcc for the first time on an M2 MacBook Air running latest Ventura with the following verbose error.  Don't think I missed a step in the setup.   Any suggestions?

```
dean@lucky13 macos % ./build.sh
python3 ../../py/makeversionhdr.py build-standard/genhdr/mpversion.h
GEN build-standard/genhdr/qstr.i.last
python3 ../../py/makeqstrdefs.py pp clang -E output build-standard/genhdr/qstr.i.last cflags -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I./SDL2.framework/Headers -I./soundio -I../../shared/readline -I../../extmod/tulip -I../../extmod/amy/src/ -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_MBEDTLS=1 -I../../lib/mbedtls/include -DMICROPY_PY_BTREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 "-I/usr/local/homebrew/opt/libffi/include" -DMICROPY_PY_FFI=1 -Og -fdata-sections -ffunction-sections -Ivariants/standard  -target x86_64-apple-macos10.15  -g -DVIRTUAL_MIDI -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  -DMICROPY_MODULE_FROZEN_STR -DMICROPY_HW_MCU_NAME="\"`sysctl -n machdep.cpu.brand_string`\"" -Wno-double-promotion -Wno-unused-function -DNO_QSTR cxxflags -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I./SDL2.framework/Headers -I./soundio -I../../shared/readline -I../../extmod/tulip -I../../extmod/amy/src/ -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_MBEDTLS=1 -I../../lib/mbedtls/include -DMICROPY_PY_BTREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 "-I/usr/local/homebrew/opt/libffi/include" -DMICROPY_PY_FFI=1 -Og -fdata-sections -ffunction-sections -Ivariants/standard -target x86_64-apple-macos10.15 -g -DVIRTUAL_MIDI -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16 -DMICROPY_MODULE_FROZEN_STR -DMICROPY_HW_MCU_NAME="\"`sysctl -n machdep.cpu.brand_string`\"" -Wno-unused-function -DNO_QSTR sources ../../lib/mbedtls/library/aes.c ../../lib/mbedtls/library/aesni.c ../../lib/mbedtls/library/arc4.c ../../lib/mbedtls/library/asn1parse.c ../../lib/mbedtls/library/asn1write.c ../../lib/mbedtls/library/base64.c ../../lib/mbedtls/library/bignum.c ../../lib/mbedtls/library/blowfish.c ../../lib/mbedtls/library/camellia.c ../../lib/mbedtls/library/ccm.c ../../lib/mbedtls/library/certs.c ../../lib/mbedtls/library/chacha20.c ../../lib/mbedtls/library/chachapoly.c ../../lib/mbedtls/library/cipher.c ../../lib/mbedtls/library/cipher_wrap.c ../../lib/mbedtls/library/cmac.c ../../lib/mbedtls/library/ctr_drbg.c ../../lib/mbedtls/library/debug.c ../../lib/mbedtls/library/des.c ../../lib/mbedtls/library/dhm.c ../../lib/mbedtls/library/ecdh.c ../../lib/mbedtls/library/ecdsa.c ../../lib/mbedtls/library/ecjpake.c ../../lib/mbedtls/library/ecp.c ../../lib/mbedtls/library/ecp_curves.c ../../lib/mbedtls/library/entropy.c ../../lib/mbedtls/library/entropy_poll.c ../../lib/mbedtls/library/error.c ../../lib/mbedtls/library/gcm.c ../../lib/mbedtls/library/havege.c ../../lib/mbedtls/library/hmac_drbg.c ../../lib/mbedtls/library/md2.c ../../lib/mbedtls/library/md4.c ../../lib/mbedtls/library/md5.c ../../lib/mbedtls/library/md.c ../../lib/mbedtls/library/md_wrap.c ../../lib/mbedtls/library/oid.c ../../lib/mbedtls/library/padlock.c ../../lib/mbedtls/library/pem.c ../../lib/mbedtls/library/pk.c ../../lib/mbedtls/library/pkcs11.c ../../lib/mbedtls/library/pkcs12.c ../../lib/mbedtls/library/pkcs5.c ../../lib/mbedtls/library/pkparse.c ../../lib/mbedtls/library/pk_wrap.c ../../lib/mbedtls/library/pkwrite.c ../../lib/mbedtls/library/platform.c ../../lib/mbedtls/library/platform_util.c ../../lib/mbedtls/library/poly1305.c ../../lib/mbedtls/library/ripemd160.c ../../lib/mbedtls/library/rsa.c ../../lib/mbedtls/library/rsa_internal.c ../../lib/mbedtls/library/sha1.c ../../lib/mbedtls/library/sha256.c ../../lib/mbedtls/library/sha512.c ../../lib/mbedtls/library/ssl_cache.c ../../lib/mbedtls/library/ssl_ciphersuites.c ../../lib/mbedtls/library/ssl_cli.c ../../lib/mbedtls/library/ssl_cookie.c ../../lib/mbedtls/library/ssl_srv.c ../../lib/mbedtls/library/ssl_ticket.c ../../lib/mbedtls/library/ssl_tls.c ../../lib/mbedtls/library/timing.c ../../lib/mbedtls/library/x509.c ../../lib/mbedtls/library/x509_create.c ../../lib/mbedtls/library/x509_crl.c ../../lib/mbedtls/library/x509_crt.c ../../lib/mbedtls/library/x509_csr.c ../../lib/mbedtls/library/x509write_crt.c ../../lib/mbedtls/library/x509write_csr.c ../../lib/mbedtls/library/xtea.c ../../extmod/modbtree.c ../../lib/berkeley-db-1.xx/btree/bt_close.c ../../lib/berkeley-db-1.xx/btree/bt_conv.c ../../lib/berkeley-db-1.xx/btree/bt_debug.c ../../lib/berkeley-db-1.xx/btree/bt_delete.c ../../lib/berkeley-db-1.xx/btree/bt_get.c ../../lib/berkeley-db-1.xx/btree/bt_open.c ../../lib/berkeley-db-1.xx/btree/bt_overflow.c ../../lib/berkeley-db-1.xx/btree/bt_page.c ../../lib/berkeley-db-1.xx/btree/bt_put.c ../../lib/berkeley-db-1.xx/btree/bt_search.c ../../lib/berkeley-db-1.xx/btree/bt_seq.c ../../lib/berkeley-db-1.xx/btree/bt_split.c ../../lib/berkeley-db-1.xx/btree/bt_utils.c ../../lib/berkeley-db-1.xx/mpool/mpool.c modtermios.c modusocket.c modffi.c ../../py/mpstate.c ../../py/malloc.c ../../py/gc.c ../../py/pystack.c ../../py/qstr.c ../../py/vstr.c ../../py/mpprint.c ../../py/unicode.c ../../py/mpz.c ../../py/reader.c ../../py/lexer.c ../../py/parse.c ../../py/scope.c ../../py/compile.c ../../py/emitcommon.c ../../py/emitbc.c ../../py/asmbase.c ../../py/asmx64.c ../../py/emitnx64.c ../../py/asmx86.c ../../py/emitnx86.c ../../py/asmthumb.c ../../py/emitnthumb.c ../../py/emitinlinethumb.c ../../py/asmarm.c ../../py/emitnarm.c ../../py/asmxtensa.c ../../py/emitnxtensa.c ../../py/emitinlinextensa.c ../../py/emitnxtensawin.c ../../py/formatfloat.c ../../py/parsenumbase.c ../../py/parsenum.c ../../py/emitglue.c ../../py/persistentcode.c ../../py/runtime.c ../../py/runtime_utils.c ../../py/scheduler.c ../../py/nativeglue.c ../../py/pairheap.c ../../py/ringbuf.c ../../py/stackctrl.c ../../py/argcheck.c ../../py/warning.c ../../py/profile.c ../../py/map.c ../../py/obj.c ../../py/objarray.c ../../py/objattrtuple.c ../../py/objbool.c ../../py/objboundmeth.c ../../py/objcell.c ../../py/objclosure.c ../../py/objcomplex.c ../../py/objdeque.c ../../py/objdict.c ../../py/objenumerate.c ../../py/objexcept.c ../../py/objfilter.c ../../py/objfloat.c ../../py/objfun.c ../../py/objgenerator.c ../../py/objgetitemiter.c ../../py/objint.c ../../py/objint_longlong.c ../../py/objint_mpz.c ../../py/objlist.c ../../py/objmap.c ../../py/objmodule.c ../../py/objobject.c ../../py/objpolyiter.c ../../py/objproperty.c ../../py/objnone.c ../../py/objnamedtuple.c ../../py/objrange.c ../../py/objreversed.c ../../py/objset.c ../../py/objsingleton.c ../../py/objslice.c ../../py/objstr.c ../../py/objstrunicode.c ../../py/objstringio.c ../../py/objtuple.c ../../py/objtype.c ../../py/objzip.c ../../py/opmethods.c ../../py/sequence.c ../../py/stream.c ../../py/binary.c ../../py/builtinimport.c ../../py/builtinevex.c ../../py/builtinhelp.c ../../py/modarray.c ../../py/modbuiltins.c ../../py/modcollections.c ../../py/modgc.c ../../py/modio.c ../../py/modmath.c ../../py/modcmath.c ../../py/modmicropython.c ../../py/modstruct.c ../../py/modsys.c ../../py/moduerrno.c ../../py/modthread.c ../../py/vm.c ../../py/bc.c ../../py/showbc.c ../../py/repl.c ../../py/smallint.c ../../py/frozenmod.c ../../extmod/moduasyncio.c ../../extmod/moductypes.c ../../extmod/modujson.c ../../extmod/moduos.c ../../extmod/modure.c ../../extmod/moduzlib.c ../../extmod/moduheapq.c ../../extmod/modutimeq.c ../../extmod/moduhashlib.c ../../extmod/moducryptolib.c ../../extmod/modubinascii.c ../../extmod/virtpin.c ../../extmod/machine_bitstream.c ../../extmod/machine_mem.c ../../extmod/machine_pinbase.c ../../extmod/machine_signal.c ../../extmod/machine_pulse.c ../../extmod/machine_pwm.c ../../extmod/machine_i2c.c ../../extmod/machine_spi.c ../../extmod/modbluetooth.c ../../extmod/modussl_axtls.c ../../extmod/modussl_mbedtls.c ../../extmod/moduplatform.c ../../extmod/modurandom.c ../../extmod/moduselect.c ../../extmod/moduwebsocket.c ../../ [... cut ...]
make: *** Deleting file `build-standard/genhdr/qstr.i.last'
error: write on a pipe with no reader
cp: tulip.i386: No such file or directory
error: write on a pipe with no reader
error: write on a pipe with no reader
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: can't open file: dev/Tulip CC.app/Contents/MacOS/tulip (No such file or directory)
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: can't open file: dev/Tulip CC.app/Contents/MacOS/tulip (No such file or directory)
```

bwhitman, 2023-02-23:
Ugh, libffi! So annoying. I thought I fixed this, but I bet my env was dirty when I declared success.

Can you try `brew install libffi` and try again for me?

blackketter, 2023-02-23:
Alas, I already had libffi installed, reinstalled it for good measure, but same behavior...  (On a whim I also tried building under Rosetta and native, no help.)

bwhitman, 2023-02-23:
Dang it, it was worth a shot. OK, I'll boot a fresh macOS VM and get this one for good (hopefully.)

blackketter, 2023-02-23:
Thanks!  Let me know if I can help...

bwhitman, 2023-02-23:
I just gave it a go and it worked on a fresh VM (by installing libffi with homebrew.) 
My process was:
- make a new ventura VM in parallels
- clone the Tulip repository (do the first run tools download to get access to `git`) 
- install homebrew using the Tulip Desktop build instructions (and add brew to the path per its instructions)
- `brew install pkg-config libffi`
- `cd tulipcc/ports/macos`
- `./build.sh`

That made a working version of Tulip Desktop for me. 
I notice that brew's `libffi` is "keg-only" and installs (for me) into `/opt/homebrew/opt/libffi` -- that's where `./build.sh` looks for it. Do you have libffi files in there, or did your homebrew put them somewhere else?

blackketter, 2023-02-24:
Apparently I had a half-arm/half-i386 version of brew running.  Full upgrade to arm and it builds fine.  Sorry for the false alarm.  THANKS!

drepetto, 2023-04-29:
Hello, I'm having a libffi problem. 

macOS Monterey 12.6.5

I installed via brew as instructed and I did the exports suggested to deal with the "keg only" install:

LDFLAGS=-L/usr/local/opt/libffi/lib
CPPFLAGS=-I/usr/local/opt/libffi/include
PKG_CONFIG_PATH=/usr/local/opt/libffi/lib/pkgconfig

But the build env can't find ffi.h: 

douglas@FickleTight-BerraTyte macos % ./build.sh
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-standard/genhdr/qstr.i.last
modffi.c:32:10: fatal error: 'ffi.h' file not found
#include <ffi.h>
         ^~~~~~~
1 error generated.
Traceback (most recent call last):
  File "/Users/douglas/Documents/tulipcc_git/tulipcc/ports/macos/../../py/makeqstrdefs.py", line 196, in <module>
    preprocess()
  File "/Users/douglas/Documents/tulipcc_git/tulipcc/ports/macos/../../py/makeqstrdefs.py", line 70, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/multiprocessing/pool.py", line 873, in next
    raise value
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/Users/douglas/Documents/tulipcc_git/tulipcc/ports/macos/../../py/makeqstrdefs.py", line 54, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/local/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['clang', '-E', '-I../../lib/berkeley-db-1.xx/PORT/include', '-I.', '-I../..', '-Ibuild-standard', '-I./SDL2.framework/Headers', '-I./soundio', '-I../../shared/readline', '-I../../extmod/tulip', '-I../../extmod/amy/src/', '-Wall', '-Werror', '-Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-Wdouble-promotion', '-Wfloat-conversion', '-std=gnu99', '-DUNIX', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DMICROPY_PY_USSL=1', '-DMICROPY_SSL_MBEDTLS=1', '-I../../lib/mbedtls/include', '-DMICROPY_PY_BTREE=1', '-DMICROPY_USE_READLINE=1', '-DMICROPY_PY_TERMIOS=1', '-DMICROPY_PY_SOCKET=1', '-DMICROPY_PY_THREAD=1', '-DMICROPY_PY_THREAD_GIL=0', '-I/usr/local/homebrew/opt/libffi/include', '-DMICROPY_PY_FFI=1', '-Og', '-fdata-sections', '-ffunction-sections', '-Ivariants/standard', '-target', 'x86_64-apple-macos10.15', '-g', '-DVIRTUAL_MIDI', '-DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool', '-DMICROPY_MODULE_FROZEN_MPY', '-DMPZ_DIG_SIZE=16', '-DMICROPY_MODULE_FROZEN_STR', '-DMICROPY_HW_MCU_NAME="Intel(R) Core(TM) i5-4278U CPU @ 2.60GHz"', '-Wno-double-promotion', '-Wno-unused-function', '-DNO_QSTR', '../../lib/berkeley-db-1.xx/btree/bt_close.c', '../../lib/berkeley-db-1.xx/btree/bt_conv.c', '../../lib/berkeley-db-1.xx/btree/bt_debug.c', '../../lib/berkeley-db-1.xx/btree/bt_delete.c', '../../lib/berkeley-db-1.xx/btree/bt_get.c', '../../lib/berkeley-db-1.xx/btree/bt_open.c', '../../lib/berkeley-db-1.xx/btree/bt_overflow.c', '../../lib/berkeley-db-1.xx/btree/bt_page.c', '../../lib/berkeley-db-1.xx/btree/bt_put.c', '../../lib/berkeley-db-1.xx/btree/bt_search.c', '../../lib/berkeley-db-1.xx/btree/bt_seq.c', '../../lib/berkeley-db-1.xx/btree/bt_split.c', '../../lib/berkeley-db-1.xx/btree/bt_utils.c', '../../lib/berkeley-db-1.xx/mpool/mpool.c', 'modtermios.c', 'modusocket.c', 'modffi.c', '../../py/mpstate.c', '../../py/malloc.c', '../../py/gc.c', '../../py/pystack.c', '../../py/qstr.c', '../../py/vstr.c', '../../py/mpprint.c', '../../py/unicode.c', '../../py/mpz.c', '../../py/reader.c', '../../py/lexer.c', '../../py/parse.c', '../../py/scope.c', '../../py/compile.c', '../../py/emitcommon.c', '../../py/emitbc.c', '../../py/asmbase.c', '../../py/asmx64.c', '../../py/emitnx64.c', '../../py/asmx86.c', '../../py/emitnx86.c', '../../py/asmthumb.c', '../../py/emitnthumb.c', '../../py/emitinlinethumb.c', '../../py/asmarm.c', '../../py/emitnarm.c', '../../py/asmxtensa.c', '../../py/emitnxtensa.c', '../../py/emitinlinextensa.c', '../../py/emitnxtensawin.c', '../../py/formatfloat.c', '../../py/parsenumbase.c', '../../py/parsenum.c', '../../py/emitglue.c', '../../py/persistentcode.c', '../../py/runtime.c', '../../py/runtime_utils.c', '../../py/scheduler.c', '../../py/nativeglue.c', '../../py/pairheap.c', '../../py/ringbuf.c', '../../py/stackctrl.c', '../../py/argcheck.c', '../../py/warning.c', '../../py/profile.c', '../../py/map.c', '../../py/obj.c', '../../py/objarray.c', '../../py/objattrtuple.c', '../../py/objbool.c', '../../py/objboundmeth.c', '../../py/objcell.c', '../../py/objclosure.c', '../../py/objcomplex.c', '../../py/objdeque.c']' returned non-zero exit status 1.
make: *** [build-standard/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build-standard/genhdr/qstr.i.last'
error: write on a pipe with no reader
error: write on a pipe with no reader
cp: tulip.i386: No such file or directory
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
[... cut ...]
error: write on a pipe with no reader
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open file: dev/Tulip CC.app/Contents/MacOS/tulip (No such file or directory)
douglas@FickleTight-BerraTyte macos % error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader
error: write on a pipe with no reader

bwhitman, 2023-04-29:
Going to reopen this to hopefully get Monterey build working for Desktop !

bwhitman, 2023-04-29:
@drepetto - i created a Monterey VM and followed the instructions and had no issue compiling/running Tulip Desktop. I'm not sure why you did the "keg only" thing, I don't think you need to, I just did default `brew install libffi`. 

Here's what I did:

* make a new Monterey VM in parallels
* clone the Tulip repository (do the first run tools download to get access to git)
* install homebrew using the Tulip Desktop build instructions (and add brew to the path per its instructions)
* brew install pkg-config libffi
* cd tulipcc/ports/macos
* ./build.sh

Maybe trash your homebrew install (or just libffi?) and try again if you can?

drepetto, 2023-04-30:
I didn't do the "keg only" on purpose, brew did that for me because it detected a native MacOS version. Then it gave the advice at the end to add exports.

bwhitman, 2023-04-30:
Right, yeah --skip / remove the exports and try again.

bwhitman, 2023-04-30:
Douglas was running Monterey on an intel (arch reports `i386`, not `x86_64`) and homebrew was in `/usr/local` not `/usr/local/homebrew` as it was on my iMac from 2020 running Monterey. I've updated that (and fixed a typo) in the Makefile for non-arm64 builds. I'll keep this open to see if others are still having this issue. Do let me know!

bwhitman, 2023-04-30:
(Also, there's never a need to do the keg-only link command it suggests -- Tulip finds the libraries explicitly.)

closed 2024-01-12 by bwhitman