shorepine/tulipcc #16: Octal flash discrepancy for WROOM-1 and WROOM-2

bwhitman, 2023-02-28:
If using a WROOM-1 dev board (like the N8R* one I suggest for the breakout or breadboard Tulip) you need to disable Octal Flash in menuconfig's Serial Flasher Config. 

For the integrated board and a WROOM-2 dev board (N32R8) Octal flash needs to be on.

This is easily fixable but people are going to run into it. I should update the instructions, and/or provide a TULIP4_OCTAL and TULIP4 board in Micropython.

bwhitman, 2023-02-28:
To help people find this issue, the error you'll see if this is you will be

`E (589) cpu_start: Octal Flash option selected, but EFUSE not configured!`

Just go and remove Octal flash from `idf.py menuconfig` in Serial Flasher settings.

bwhitman, 2023-03-01:
I created a new board type and updated the docs.

bwhitman, 2023-03-12:
Re-opening this as a user in the wild is having trouble booting their tulip board. I am debugging

bwhitman, 2023-03-13:
I removed the N32R8 board type as it was confusing.

I now just have a note to do the (one time) change to the octal flash if you're using the N8R8. 

I think this is easier and would assume over time most people to use the WROOM-2 or N32R8 chips.

bwhitman, 2023-03-13:
From here

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html

It does seem that this is a misconfiguration for N8R8 boards. I can try burning the fuse on mine and see if it helps

bwhitman, 2023-03-13:
Burning the fuse looked like it was gonna work
```
Connecting....
Detecting chip type... ESP32-S3
espefuse.py v3.2-dev
The efuses to burn:
  from BLOCK0
     - FLASH_TYPE

Burning efuses:

    - 'FLASH_TYPE' (Selects SPI flash type) 0b0 -> 0b1

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
	(written ): 0x0000000080000100000000000000d1f50000000000000000
	(to write): 0x000000000000020000000000000000000000000000000000
	(coding scheme = NONE)
. 
This is an irreversible operation!
BURN BLOCK0  - OK (all write block bits are set)
Reading updated efuses...
Checking efuses...
Successful
```

but then

```
E (704) esp_image: Checksum failed. Calculated 0x99 read 0x76
E (704) boot: Factory app partition is not bootable
E (705) boot: No bootable app partitions in the partition table
```

on boot... strange. I've asked about this on the espressif forums

bwhitman, 2023-03-14:
Oh, the N8R8 rev 1 doesn't have octal flash. It has octal **SPI**, just for the PSRAM. So the flash access is slower on the N8R8. It'll still work on tulip, but the write/read from flash will be slower. There is a later rev of the N8R8 that does have octal flash. The N16R8 does have octal flash. The N32R8 does have octal flash. Whoosh. 

https://www.esp32.com/viewtopic.php?f=12&t=32614

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

bwhitman, 2023-03-14:
I think trying to work around this just for the rev1 N8R8 is probably silly. I'll make the docs clear that if you see the EFUSE error, to just turn off octal flash. And I'll update the supported boards/modules list to include the N16 and N8R8 rev 2.

jd3096-mpy, 2023-03-15:
I update your lastest reop and complie fireware in ubuntu,finally it works ,thanks for your support!!!
[image]
My board is N8R8,screen is 5-incn 800x480 rgb.
I am planning to make a small micropython laptop,so I will design new PCB board.
Next step I will trying to link keyboard (My board dosen't have USB port ,only UART port).
Is the screen color correct? BG color looks like green,it's different from your pics.

bwhitman, 2023-03-15:
@jd3096-mpy that's great!!! 
it's hard to tell the color from the pic. but yes, mine looks "bluer". It could be a color issue on the display or maybe a mis-wired color pin? Try `tulip.bg_png("ex/g/colorbars.png",0,0)` and see if that looks ok?

jd3096-mpy, 2023-03-26:
> tulip.bg_png("ex/g/colorbars.png",0,0)

[image]
It seems ok?
And I have two questions.
1 Usbkeyboard don't work,now I use uart0 to sending commands,I use an old usb keyboard in my office,linked gpio 19 20 5v gnd ,even the caps lock  don't light on.
2 How to transfer files from computer to board? I treid micropython IDE like Thonny,but it dosen't work.

commit e02dfa56 (2023-03-01): new board type, to fix #16
closed 2023-04-03 by bwhitman