shorepine/tulipcc #20: I try to build firmware by it starts failed

jd3096-mpy, 2023-03-11:
uart print:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0xe34
load:0x403b6000,len:0x9e8
load:0x403ba000,len:0x2be4
SHA-256 comparison failed:
Calculated: 3356bfb3cf5cdc8b9f12bd72c6d62aa0a5824e33f4c54cc5d72bbd54d9a0ad56
Expected: f8562e64c0a43395093d8bab1f4be4ea2db25043e3be49520448dc36ff4b5ca7
Attempting to boot anyway...
entry 0x403b61d4
Starting display on core 0
setting duty to 4000
Starting Alles on core 1
Starting MicroPython on core 1
St�.�I'm renderer #1 on core #0 and i'm handling oscs 32 up until 64

screen it reboot 

what's the problem? Thanks

bwhitman, 2023-03-11:
The sha error seems to be ignorable ,  but I've never seen it. Can you explain more why you don't think it's working? The rest of the monitor output looks right.

jd3096-mpy, 2023-03-11:
> The sha error seems to be ignorable , but I've never seen it. Can you explain more why you don't think it's working? The rest of the monitor output looks right.
Thanks for your reply, I am very intertsted in this reop,but maybe I still don't understand the way it works.
I use ubuntu to make the firmware, the way as the micropython offical repo recommend.
So I get the firmware.bin, this should be write to addr 0x0.
but in release, it says
parttool.py write_partition --partition-name=vfs --input=tulip-lfs.bin to write the tulip FS, and
parttool.py write_partition --partition-name=factory --input=micropython.bin to write the Tulip/MP firmware.

where can I get tulip-lfs.bin? And what's the addr it shoud be write?Thanks.

bwhitman, 2023-03-11:
See tulip_fs_create.py in ports/esp32s3. That both creates and flashes the Tulip file system (not the firmware).

jd3096-mpy, 2023-03-12:
> See tulip_fs_create.py in ports/esp32s3. That both creates and flashes the Tulip file system (not the firmware).

I use mac to follow your every steps,but the same result:build firmware success,run failed.
I see your board is N8R8,I only have N8R2 and N16R8,they both failed,Is that the reason I run failed?
I buy a N8R8 board,I will try the I got it.

In https://github.com/bwhitman/tulipcc/blob/main/ports/esp32s3/boards/TULIP4_N32R8/mpconfigboard.cmake
set(SDKCONFIG_DEFAULTS
    boards/sdkconfig.base
    boards/sdkconfig.usb
    boards/sdkconfig.spiram_sx
    boards/TULIP4/sdkconfig.board
)
the same setting as N8R8?
or shoule be like this
set(SDKCONFIG_DEFAULTS
    boards/sdkconfig.base
    boards/sdkconfig.usb
    boards/sdkconfig.spiram_sx
    boards/TULIP4_N32R8/sdkconfig.board
)
Is that a mistake?

bwhitman, 2023-03-13:
Yes, thanks @jd3096-mpy . That is a mistake and for now we're tracking that issue over at #16 . I'm not certain we need the separate board types. For now, if you update you should be able to just use TULIP4 as the board type on an N8R8 or N16R8. Your N8R2 will not work, we require 8MB of ram and octal flash.

bwhitman, 2023-03-13:
Going to close this issue as we're chatting over on the discussions page and we'll use #16 to figure out the board type issue.

closed 2023-03-13 by bwhitman