shorepine/tulipcc #53: About resolution

jd3096-mpy, 2023-04-03:
Form api docs,I know that I can change resolution with this api
`tulip.timing(1024, 600, 256, 150, 139, 140, 20, 20, 12, 20)`
I found that when I set height between 240 and 300,screen shows error like this,many black lines and diveid to three parts:
[image]

How to fix it?Thanks.

bwhitman, 2023-04-03:
are you using our 10.1" screen or something custom? It looks smaller than ours. 

One thing to keep in mind is the screen Y size needs to be a multiple of the BOUNCE_BUFFER_PX variable. 

Without having your screen, it'd be hard for me to debug. But there's no reason Tulip can't work at heights between 240 and 300.

jd3096-mpy, 2023-04-03:
Oh,I test it and found that height must be the Multiples of 12!
Because the font is 8x12
height 300,288,276,both ok
I shoud use 480x276  resolution though my screen is 480x272

closed 2023-04-03 by jd3096-mpy