shorepine/tulipcc #103: Desktop default audio device

bwhitman, 2023-06-27:
You can set audio device by running the app like `./dev/Tulip\ CC.app/Contents/MacOS/tulip -d 2` but on double clicking it'll use device `-1`. For me, at least this is never the device i have set as my default. I don't know why.

bwhitman, 2024-06-21:
This is now even weirder for me, after a macOS update -- I can no longer even set the device to anything but the system default -- not even after changing the system default (in sound settings) ! and `-d X` no longer has any effect. I've tried this on two macOS machines and both exhibit the same behavior. I may have some misconfiguration but I'll look to see if maybe miniaudio has an update.

bwhitman, 2024-06-21:
(this is not a tulip specific issue - it will appear in AMY and Alles as well, but I'll track it here)

jminor, 2024-08-19:
I downloaded Tulip Desktop (August 2024 release) and found that it produces no sound on my MacBook Pro (macOS 14.5).

After finding this issue, I tried running `~/Downloads/Tulip\ Desktop.app/Contents/MacOS/tulip -d 1` from the Terminal and now it works!

Is there a way to specify the audio device from `boot.py`?

bwhitman, 2024-08-19:
Unfortunately not yet. The audio device is setup in C before even micropython boots. It would be possible to restart the audio device on Tulip Desktop in Python using another device ID, so I'll take a look at that.

bwhitman, 2024-08-19:
@jminor - do help us out, do you have any other audio devices connected to your computer (this could include things like HDMI monitors, or cameras) , and is your macbook's speakers set as the default audio device? (The blue icon when you click on the volume icon in the menu bar.) I think this may be a `miniaudio` bug but i'd love to collect data about it not working.

jminor, 2024-08-19:
I don't have any physical audio devices attached, but there are a couple of virtual ones.

`tulip -l` shows:
```
0 - BlackHole 16ch
1 - MacBook Pro Speakers
2 - ZoomAudioDevice
3 - Multi-Output Speakers + Blackhole 16
```

My default in System Settings is "MacBook Pro Speakers"

jminor, 2024-08-19:
More info:
System Settings shows a slightly different list of devices:
[image]

Let me know if there's anything else I can try/gather to help out. Maybe I can run a different miniaudio program?