shorepine/tulipcc #111: Linux build instructions?

znmeb, 2023-07-05:
I can probably help out with this. I might even be able to get it running on Windows via Windows Subsystem for Linux, but I can't make any guarantees about audio performance or USB connectivity.

bwhitman, 2023-07-05:
That would be great. Keyboard input is handled by SDL on Desktop builds so it should just work if you get that going.

znmeb, 2023-07-05:
I took a look at the macOS Makefile yesterday - it looks very Mac-specific, I'm looking into some more generic build configuration tools like CMake, unless there's something simpler that will work. CMake would in theory give you a single source tree that would build on Linux, Windows and Mac. But I don't own a Mac so I couldn't test that part.

bwhitman, 2023-07-05:
For a head start on a linux build of Tulip Desktop, i'd look at the [`unix` Makefile in Micropython](https://github.com/micropython/micropython/blob/master/ports/unix/Makefile)

znmeb, 2023-07-05:
> For a head start on a linux build of Tulip Desktop, i'd look at the [`unix` Makefile in Micropython](https://github.com/micropython/micropython/blob/master/ports/unix/Makefile)

It would be faster for me to buy a Mac Mini, a keyboard and a display than to learn the MicroPython build configuration process.

bwhitman, 2023-07-07:
We have updated [the Tulip Desktop page](https://github.com/bwhitman/tulipcc/blob/main/docs/tulip_desktop.md) with instructions on how to build on Linux. Haven't tested it beyond my Ubuntu 22 VM. Give it a go and we can follow up with any more issues!

closed 2023-07-07 by bwhitman