shorepine/tulipcc #117: virtualmidi.m not working after MIDI refactor

bwhitman, 2023-07-06:
It's breaking the build on macos on my machine with 
```
virtualmidi.m:39:27: error: array type 'uint8_t[8]' (aka 'unsigned char[8]') is not assignable
            last_midi_len = 0;
            ~~~~~~~~~~~~~ ^
virtualmidi.m:63:46: error: array type 'uint8_t[18]' (aka 'unsigned char[18]') is not assignable
                                last_midi[0] = bytes[2];
                                ~~~~~~~~~~~~ ^
virtualmidi.m:64:46: error: array type 'uint8_t[18]' (aka 'unsigned char[18]') is not assignable
                                last_midi[1] = bytes[1];
                                ~~~~~~~~~~~~ ^
virtualmidi.m:65:46: error: array type 'uint8_t[18]' (aka 'unsigned char[18]') is not assignable
                                last_midi[2] = bytes[0];
                                ~~~~~~~~~~~~ ^
virtualmidi.m:66:47: error: array type 'uint8_t[8]' (aka 'unsigned char[8]') is not assignable
                                last_midi_len = 3;
                                ~~~~~~~~~~~~~ ^
```

@dpwe any idea?

dpwe, 2023-07-06:
Yes, idea.  I'll fix it.

dpwe, 2023-07-06:
Try https://github.com/bwhitman/tulipcc/pull/118

closed 2023-07-06 by bwhitman