shorepine/tulipcc #128: `editor.c` - we need soft wrapping or h-scroll

bwhitman, 2023-07-26:
the current editor gleefully will just trim the right side of a line if it goes over `TFB_COLS`. (not in the file, just on screen.) there's no way to see it or edit it. we have two choices

 (1) allow for horizontal wrapping: we allow users to move off the right side of the screen, the text on that line will "scroll" to the right , we place a special character at the start of the line to show we're not at the start of the line

 (2) soft wrap the longer lines -- we show a line counter on the left side of the editor. if a line is longer than `TFB_COLS` we draw the remainder on the next line but dont' show a new line number. Arrowing-down/up jumps actual file lines, not screen lines (I think)

coolcoder613eb, 2024-05-02:
This is important to me, also toggling syntax highlighting, and better keybindings, and handling cr/lf endings.
Also, you can't use backspace when entering the save filename, which leads to files that are hard to remove, necessitating something like `rm(os.listdir()[4])`.
Also `rm` should be more like this:
```Python
def rm(d):  # Remove file or tree
    try:
        if os.stat(d)[0] & 0x4000:  # Dir
            for f in os.ilistdir(d):
                if f[0] not in ('.', '..'):
                    rm("/".join((d, f[0])))  # File or Dir
            os.rmdir(d)
        else:  # File
            os.remove(d)
    except:
        print("rm of '%s' failed" % d)
```
To allow removing non-empty directories.
Hmm... I'm getting a bit off-topic.

bwhitman, 2024-05-25:
@coolcoder613eb i've fixed the editor save backspace situation, thanks for pointing that out.

bwhitman, 2024-06-11:
I'm going to punt on h-scroll until we port the editor to pure python (using the TFB). it will help with a lot more than just this issue. 
for now, the editor works well enough for most purposes.