shorepine/tulipcc #141: Alles mesh crashes local audio playback

bwhitman, 2023-09-19:
It's unclear if we want to support it (or if we can?) but currently, a Tulip in Alles mesh mode (`alles.mesh()` after joining wifi) will also play locally, i.e. it's its own Alles node as well as controlling a mesh. We set latency to 1000ms and it will receive its own multicast messages immediately for playback. 

Simple uses of this work fine but sending more than a few messages a second (like `alles.drums()`) will eventually (a few seconds) cause the "host" Tulip to click and stop rendering audio. It still sends messages to the mesh. It still receives them locally and attempts to play them locally. And sometimes it'll come back after a long time (many seconds.) But something in the rendering or i2s fill-buffer stage is no longer making audio go out the i2s port on Tulip for long stretches of time.

I haven't been able to find the issue on a quick printf debug session. A solution could be we just disable local playback when Tulip is in mesh mode. But it may be a symptom of a different issue so i'd like to track it down.

bwhitman, 2023-09-19:
The click/silence issue is that NaNs were getting into the renderers. I _think_ the NaNs are happening because something is stealing the CPU from the render() function midway through. 

Increasing priority of `ALLES_RENDER_TASK_PRIORITY` and  `ALLES_FILL_BUFFER_TASK_PRIORITY` to `(ESP_TASK_PRIO_MAX - 1)` is fixing this for now. I'm able to have tulip send `alles.drums()` out to a mesh and also play it back in sync itself as a node. Everything is solid. 

We had dealt with something similar during testing of @dpwe's `xanadu.py`, and I had increased those priorities then to max-5. But i guess wifi being on is stealing CPU from the render tasks while they're running. 
More testing needed, but i'll commit the priority bump, and think about a flag for `alles.mesh()` like `local_node=True` to indicate if the Tulip should play as well.

In short - we're pushing the esp to the limit and I think we can squeeze in playback + multicast controller but it may get annoying later with more complicated things.

nvillar, 2023-09-19:
Very cool, thank you for the fix. A local_node flag makes a lot of sense to me - when using a Tulip as a controller for an Alles Mesh you're less likely to need the local rendering and would be great to free up the cycles.

closed 2023-10-02 by bwhitman