shorepine/tulipcc #145: Unable to compile for T-Deck

kikura, 2023-10-10:
Can anyone provide compiled file to flash as unable to successfully compile. Various errors and crashes on Linux distro s running from USB. Only have Windows PC.

Thanks in advance.

kikura, 2023-10-11:
> Sure, here: tdeck.zip
> I really do recommend trying to figure out your build challenges. Tulip is fast changing and we don't recommend (or give support for) uploading binaries directly. Can you paste in any errors / crashes you're seeing on Linux trying to build and flash Tulip?

Thank-you so much. Looking forward to try this out 👍

Nick.

kikura, 2023-10-29:
Okay, I have tried the flashing instructions on the latest version of Ubuntu.
I got as far as:

esp-idf-v5.1-rc2/install.sh esp32s3

but get the following output and error message:

Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32s3
Installing tools: xtensa-esp-elf-gdb, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Installing xtensa-esp-elf-gdb@12.1_20221002
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz
Destination: /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
80%WARNING: file size mismatch for /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp, expected 32006939, got 25735652
WARNING: Download failure: <urlopen error retrieval incomplete: got only 25735652 out of 32006939 bytes>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz to /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz
Destination: /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
89%WARNING: file size mismatch for /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp, expected 32006939, got 28766692
WARNING: Download failure: <urlopen error retrieval incomplete: got only 28766692 out of 32006939 bytes>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz to /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz
Destination: /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
92%WARNING: file size mismatch for /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp, expected 32006939, got 29721336
WARNING: Download failure: <urlopen error retrieval incomplete: got only 29721336 out of 32006939 bytes>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v12.1_20221002/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz to /home/nick/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-x86_64-linux-gnu.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Please make sure you have a working Internet connection.

bwhitman, 2023-10-29:
@kikura if that keeps happening when you try again, I'd ask for help at the esp-idf project https://github.com/espressif/esp-idf/issues

closed 2023-12-15 by bwhitman