shorepine/tulipcc #156: Update esp Tulip for later AMY

bwhitman, 2024-01-06:
I broke the esp32s3 build by mistakenly merging in the latest AMY without first fixing up how AMY is instantiated on tulip. 

I'll track that work here, I'll need to test on hardware.

For now, if you want to use esp32s3 Tulip, roll back a few commits:

```
git checkout 793324c
rm .submodules_ok
cd tulip/esp32s3/
idf.py ...
```

cc @norahqueen

bwhitman, 2024-01-06:
OK, I've committed a fix for this to main, please pull (and rm .submodules_ok), and try to build again. I confirmed that the build completes for me on esp32s3 and tdeck. I have _not_ confirmed Alles/AMY works on boot on Tdeck or bare esp32s3, as I'm away from my hardware right now. Can double check in a day or two!

JustDr00py, 2024-01-06:
I have tested the recent commit on my tdeck, everything compiles and flashes but alles.drums() now plays high pitched tones and sounds broken.

bwhitman, 2024-01-06:
That could be an older version of alles.py. Can you try to do an idf.py fullclean then flash again?

bwhitman, 2024-01-07:
Haha nope, I got my T-Deck and hear what you're hearing. Will look into it

bwhitman, 2024-01-07:
We have two issues, one of which I fixed 

(a) we were sending stereo samples to a mono-configured i2s chip on the Tdeck. I fixed that.
(b) Something about the new filter modes in newer AMY make that little speaker go mad. I asked @dpwe to check it out. If you comment out the "bassline" in `alles.py` you'll hear the drum loop just fine. 

I think this is Tdeck only, but I'll test on big Tulip asap.

JustDr00py, 2024-01-07:
thank you again its working now

norahqueen, 2024-01-07:
Got it working on my tdeck, thank you @bwhitman 

Can confirm getting the same issue with `alles.drums()`

bwhitman, 2024-01-09:
@JustDr00py @norahqueen @dpwe fixed the filter on AMY that was making the T-Deck go wild. Pull up to main and `rm .submodules_ok`, build again and it should be fine!

closed 2024-01-09 by bwhitman