shorepine/tulipcc #160: Tulip CC v4r10

bwhitman, 2024-01-10:
We're working on a version of Tulip that (hopefully, no promises yet) anyone can buy with a screen attached. It's based on the [MaTouch 7" from Makerfabs](https://wiki.makerfabs.com/MaTouch_ESP32_S3_7_Paralle_%20TFT_with_Touch.html), and Makerfabs are helping us with the design. The MaTouch 7" is _almost_ a usable Tulip on its own. It uses a mono speaker instead of line-out audio, doesn't have hardware MIDI, and you can't use USB at the same time as audio (they share pins.) And the MCU (N16R8) doesn't support the fast 120MHz RAM access speeds. But the screen, USB host, battery IC are all pretty great for Tulip. We will also canonicalize nicoVR's r9 as the "DIY" tulip board for those that want to build their own and deprecate my older designs (all which have various issues.)

I'll use this issue to track the progress first getting Tulip on the existing MaTouch 7" 1024x600, then the dev and design work with makerfabs getting a Tulip board made.

For now, the MaTouch bringup is all in `tulip/tulipcc_r10` - use the usual `idf.py flash` in there to build for the MaTouch. It defaults to `N16R8`, when we have TulipR10s in hand we can also support the N32R8 120MHz mode. 

MaTouch bringup:

- [ ] Bring up screen, python
- [ ] Touchscreen (GT911)
- [ ] Audio (MAX987 mono speaker)
- [ ] USB host
- [ ] (Maybe) test MIDI by running wires from the MCU
- [ ] Test grove / I2C jack with MP 

Tulip R10 design:

- [ ] Swap MCU to ESP32-S3-WROOM-2 N32R8
- [ ] PCM5101A line-out 
- [ ] Hardware MIDI with nicoVR circuit
- [ ] Jack placement - USB host, USB serial / charging, MIDI in, MIDI out, audio out, grove/i2c, battery
- [ ] Remove SD card
- [ ] Remove GPIO header
- [ ] Different pins for USB host & I2S
- [ ] Test "high color" RGB555 slower mode

bwhitman, 2024-02-09:
A small note - if you're using the MaTouch7 as a Tulip, i recommend getting an SD card breakout (or something similar) and wiring it up for I2S (using IO11,12,13). Then you can use audio and USB at the same time and bring your own I2S breakout board.

bwhitman, 2024-03-03:
We have a rev of Tulip from Makerfabs in hand. They really work fast and did great work. There are a few issues we'll need to fix / design another rev for. I'll pull together the issues here and get it out to them asap for another round.

- [ ] I2S/USB pins were incorrectly placed, and I missed this on verification (already tested with bodge)
- [ ] 3.5mm audio jack is noisy (CPU whine) when running with USB-TTL power. It sounds great on battery alone. Move the jack and DAC to the other side of the board, and ask for any other advice to minimize noise. 
- [ ] USB-NATIVE (aka keyboard) is not being powered by battery when running only on battery as there's no 5V. Proposed solution: Use LED backlight boost circuit (AVDD10), or add a new PAM2401 or similar 
- [ ] We can't seem to adjust brightness with PWM, and we could with the MaTouch. this may be software, still looking into it
- [ ] CHRG and DONE LEDs are distracting and not needed , remove them. We get charge status from software now. 

See their schematic: 
[tulip-matouch-eagle-7.pdf](https://github.com/bwhitman/tulipcc/files/14474488/tulip-matouch-eagle-7.pdf)

bwhitman, 2024-03-03:
Here's a picture! Look at that great screen and all those ports on the side. I'm running a battery rundown test on Tulip. 

[image]

bwhitman, 2024-03-13:
OK, rev2 of Tulip v4r10 (one day we'll get naming right) is sent off to makerfabs. Fix list is:

- [ ] fix i2s pins to IO5, IO8, tie B3 and G2 to ground (tulip can eventually write RGB554 color if we want) 
- [ ] Ferrite bead on VBUS for USB-TTL
- [ ] Remove CHRG, DONE and POWER LEDs -- not needed, the CHRG one flashes annoyingly, and there will be backplate that covers them all up anyway
- [ ] 56K pullups on "USB-NATIVE"
- [ ] Solution for 5V on battery power on "USB-NATIVE" (either use AVDD10 or do another boost circuit)
- [ ] Flash Tulip bins at factory with `TULIP4_R10-full.bin` 
- [ ] Silkscreen: change "USB-NATIVE" to "USB-KEY", "J2" to "I2C", clean up CONNs

bwhitman, 2024-04-01:
Latest rev (Tulip4R10VB?) just arrived. I'll go over it and confirm -- 
[tulip-matouch-eagle-7B.PDF](https://github.com/bwhitman/tulipcc/files/14823175/tulip-matouch-eagle-7B.PDF)

bwhitman, 2024-04-04:
I confirmed the changes on the schematic and am getting 3 eval boards as soon as they can manufacture them. 

[tulip-matouch-eagle-7b.zip](https://github.com/bwhitman/tulipcc/files/14872295/tulip-matouch-eagle-7b.zip)

bwhitman, 2024-04-25:
I have my new tulip in hand! Everything seems to work great. There is a new flicker when using a keyboard with a battery built in (like a bluetooth keyboard with LEDs), probably due to the higher current draw . But with a "dumb" keyboard everything is 100%. I'll look into mitigating the flicker for the final run. I'll keep testing over the next few days but everything else looks and sounds great. 

[image]

jacquesgroenewald, 2024-04-27:
Awesome!

Got my 11 inch up and running. (had to desolder the FPC adapter and solder a connector on the other side and flip over for it to work - thankfully didnt get damaged when I had it plugged in the 1st time).

Will this be an off the shelf item from Makerfabs? If yes how much do you think it would be sold for?

bwhitman, 2024-04-27:
@jacquesgroenewald cool! What 11" display are you using? 
Yes, the idea is that anyone will be able to buy this from the Makerfabs store. 
We don't yet know the final price as we don't have a final BOM. Should know soon.

bwhitman, 2024-04-29:
We've asked Makerfabs to do a (perhaps final!) rev with a new boost chip for USB-KB. They're also fabricating an acrylic back cover, like the MaTouch has. This protects the components on the back and has cutouts for the battery and ports.

jacquesgroenewald, 2024-04-29:
The 11 inch you have specced in the guide.

I will try and take Pic later this week and place next to your guide's
picture to show what the orientation issue was.

bwhitman, 2024-05-28:
It looks like the Tulip CC board (now r11) is ready! I'll test it here later this week. Some nice marketing shots sent over from Makerfabs, you can see the acrylic back case:

[image]
[image]

bwhitman, 2024-05-30:
I got my final Tulipv4r11! The USB-KB issue is fixed. The back plexi is nice. A lipo battery can fit right between the plexi and PCB. Running it through its paces but i think we can finally call this done. More news ASAP on getting it in the store.

jacquesgroenewald, 2024-05-31:
Looking forward to see it in action!

Jacques

bwhitman, 2024-06-03:
The board is now available to buy: https://www.makerfabs.com/the-tulip-creative-computer.html

We'll do a bigger launch in a few weeks or so, but if you see this message, go ahead and get one!

closed 2024-06-03 by bwhitman