shorepine/tulipcc #193: Design spec for music subsystem

bwhitman, 2024-03-17:
Specification for planned "music subsystem" for comment:

# Music system

We now ship an "always on" music subsystem in Tulip. 

## Music voice mapping
By default, at power on, Tulip will respond to MIDI note messages and assign them to different voices in AMY, with polyphony and voice stealing. This will happen with no other apps running, and allows Tulip to just act as a synthesizer with no user interaction. For example, by default, MIDI notes on channel 0 will map to Juno-6 patch 0, channel 1 to Juno-6 patch 1, channel 2 to DX-7 patch 0, etc. 

You can adjust which voices are sent with `tulip.music_map(channel, patch_number, voice_count)`.  The channel is a MIDI channel (we use 0-15 indexing), the patch_number is an AMY patch number, and voice_count is the number of voices (polyphony) you want to support for that channel and patch. (A good rule of thumb is Tulip CC can support about 12 simultaneous total voices for Juno-6 and DX7, and 20-30 total voices for PCM and more for other simpler oscillator patches.) 

You can also send AMY patch commands (strings) with `tulip.music_map(channel, patch_string, voice_count)`. This lets your UI apps adjust patches and send updates to AMY for the synthesizer. (You can use AMY’s record_patch functionality to get a patch string, or have your app generate one.) 

These mappings will get reset to default on boot. If you want to save them, put `tulip.music_map()` commands in your boot.py. 

## Music clock
The music subsystem allows any program you write or the ones we ship to share a common music clock, so that (for example) a drum machine can stay "in time" with an arpeggiator. 

To use the clock in your code, you should first register on the music callback with `tulip.music_callback_add(my_callback, 6)`. You can remove your callback with `tulip.music_callback_remove(my_callback)`. 

When adding a callback, the second parameter (“6”) is a divider on the system level parts-per-quarter timer (currently defaults at 48). This would call your function my_callback every 6th "tick", so 8 times a quarter note. Set this to 48 (or, `tulip.ppq()`) to receive a message once a beat.

By default, your callback will receive a message 50 milliseconds ahead of the time of the intended tick, with the parameters `my_callback(tick_number, intended_time_ms)`. This is so that you can take extra CPU time to prepare to send messages at the precise time, using AMY scheduling commands, to keep in perfect sync. You can set this "lookahead" globally if you want more or less latency with tulip.music_latency(X) or get it with `tulip.music_latency()`. 

You can set the system-wide BPM (quarters per minute) with `tulip.bpm(120)` or retrieve it with `tulip.bpm()`. You can change the PPQ with `tulip.ppq(new_value)` or retrieve it with `tulip.ppq()`. You can restart the tick counter with `tulip.music_restart()`, pause it with `tulip.music_pause()` and start it with `tulip.music_start()`.

dpwe, 2024-03-18:
Love it.

bwhitman, 2024-03-18:
An addition around CCs:

By default, Tulip will support a mapping from MIDI CC messages (per channel) to the patch loaded on each channel. We'll start with using the standards from the JU-06: https://midi.guide/d/roland/ju-06/ . We'll map as much as we can in Python on Tulip to AMY commands for the three types of patches (DX7 & Juno & random). 

We'll also support remapping these CC #s for people that have MIDI controllers that send arbitrary CCs. This would look like `tulip.music_cc_map(original_cc, my_cc)`.

dpwe, 2024-03-18:
I'm less clear about how CCs would work. There's a significant amount of idiosyncratic logic required to map a MIDI control into AMY messages. For instance, the Attack slider for a Juno ADSR has to go through an Attack-specific nonlinear mapping (from MIDI value to attack time in milliseconds), then it must adjust both the attack and decay end times and rewrite the entire breakpoint strings for the VCF envelope and the amplitude envelope too (unless VCA gate is on) for each of the ramp, pulse, subosc, and noise oscillators for all the juno voices. It's not something that's easily parameterized, it ends up a nontrivial custom function for every control.  There's a lot of code handling this in Juno.py.

bwhitman, 2024-03-18:
Right, I was just gonna have it import juno.py and do it that way. The point is to be able to use CCs when e.g. juno_ui is not running. And we can one day make a similar dx7 thing.

dpwe, 2024-03-18:
Hmm, that should work for now, but only because of the synchronization between `juno.py` and `src/patches.h`.  This is enforced by the build mechanics, but feels a bit shaky - anyone can just edit juno.py, and the compiled-on patches won't necessarily stay in-sync.  

Put another way, the patches are in the KERNEL, whereas `juno.py` is in USERLAND.

dpwe, 2024-03-18:
I am personally OK with having CCs work only when something like `juno_ui` is running.  I think plug-and-play MIDI input without CCs covers most use-cases.

bwhitman, 2024-03-18:
Alright, we'll ignore CCs for now and maybe tackle it later if people want it 

So the always-on MIDI mapping will do note-ons, note-offs, pitch bend, program changes (to set voices) and that's it for now.

bwhitman, 2024-03-23:
Some updates to the clock part of this (in the branch `music` for now). This is working on Desktop and there's a demo on World called `seq.py`. 

## Music clock
The music subsystem allows any program you write or the ones we ship to share a common music clock, so that (for example) a drum machine can stay "in time" with an arpeggiator. 

To use the clock in your code, you should first register on the music callback with `tulip.seq_add_callback(my_callback)`. You can remove your callback with `tulip.seq_remove_callback(my_callback)`.  You can remove all callbacks with `tulip.seq_remove_callbacks()`. We support up to 4 callbacks running at once. 

When adding a callback, there's an optional second parameter to denote a divider on the system level parts-per-quarter timer (currently defaults at 48). If you run `tulip.seq_add_callback(my_callback, 6)`, it would call your function `my_callback` every 6th "tick", so 8 times a quarter note at a PPQ of 48. The default divider is `tulip.seq_ppq()`, so if you don't set it, your callback will activate once a quarter note. 

By default, your callback will receive a message 50 milliseconds ahead of the time of the intended tick, with the parameters `my_callback(intended_time_ms)`. This is so that you can take extra CPU time to prepare to send messages at the precise time, using AMY scheduling commands, to keep in perfect sync. You can set this "lookahead" globally for all callbacks if you want more or less latency with tulip.seq_latency(X) or get it with `tulip.seq_latency()`. 

You can set the system-wide BPM (beats, or quarters per minute) with `tulip.seq_bpm(120)` or retrieve it with `tulip.seq_bpm()`. You can change the PPQ with `tulip.seq_ppq(new_value)` or retrieve it with `tulip.seq_ppq()`. You can stop the sequencer with `tulip.seq_stop()`, and restart it with `tulip.seq_start()`.

closed 2024-03-31 by bwhitman