shorepine/tulipcc #205: SD card on T-Deck

coolcoder613eb, 2024-05-02:
If SD card slot on the T-Deck is supported, please tell me how to use it, if not, please add support for it.

bwhitman, 2024-05-02:
SD card support in Tulip. is not currently tested on T-Deck. It's my understanding that all the pieces are there to access the SD card, and may even be possible with pure Python (and no changes to Tulip itself.) Have you tried the instructions here: https://docs.micropython.org/en/latest/library/machine.SDCard.html ? The pins seem to be `SD_CS` = `GPIO39`, `MOSI` = `GPIO41`, `CLK` = `GPIO40`, `MISO` = `GPIO38`.

coolcoder613eb, 2024-05-02:
`machine.SDCard`  is not available in TulipCC.

bwhitman, 2024-05-03:
Ah, i see. I've gotten SDCard to appear in Tulip on T-Deck only by changing an #ifdef in the micropython setup. But the card gives an error on init. I'll take a deeper look at this later:

```
>>> import machine
>>> m = machine.SDCard(sck=40,miso=38,mosi=41,cs=39)
>>> uos.mount(m, '/sd')
E (187697) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16
```