shorepine/tulipcc #224: Juno6 onscreen parameter update glitches

dpwe, 2024-06-02:
When adjusting the VCF cutoff frequency using the juno6.py on-screen slider, the sound of a currently-active Juno voice glitches unpleasantly - it sounds as if the filter is fully opened up before being set with the new values.  (This is when running on Tulip hardware).

I don't think it used to do this and I don't think it's necessary. It may be a side effect of the mod to only run the VCF on all the oscillators added together instead of per-osc:  [Amy #221496d](https://github.com/bwhitman/amy/commit/221496d38be4cd183259913b979686f9cbceca72)

I want to look at the exact sequence of messages sent to Amy during these updates and look through what they would be expected to do.

dpwe, 2024-06-02:
I can't hear any glitches when I move sliders on the Tulip Desktop version.  That may be because it's just a much faster computer, or maybe it's related to multithreading on the ESP32.

bwhitman, 2024-06-04:
Dan, this currently works for me, i think from your changes to `juno.py` in AMY a couple of days ago. Are we OK to close now?

dpwe, 2024-06-04:
Yes, fixed in [Amy #8b88765](https://github.com/bwhitman/amy/commit/8b887659d9cafb234fd5b7cb410d0e811713a72d).  After the mod to make the VCF be applied once to all the oscs in a chain, I had changed the way `juno.py` updated the 4 source oscillators to update osc 0, remove its filter, clone to oscs 1, 2, 3 (so they correctly have no filter), then re-introduce the filter on osc 0.  On hardware, this process can be interrupted to generate some waveform under the partially-complete update parameters (so without the filter on osc 0), leading to the glitches.  I changed it to keep the filter in place during cloning oscs, then explicitly remove the extra filter on each of oscs 1, 2, 3.  This will still give anomalous results when output frames are generated mid-update, but at least they won't be super bright and loud.

I actually don't think there's such a strong need for updating juno oscs using `clone_osc` any more, the load on the `Amy.send` channel has mostly been relieved by the `voices` mechanism.  In the future, I may rework the `juno.py` updates to write each osc independently, and also to allow VCF updates only to touch osc 0.  That would ideally remove many param-change glitches.

closed 2024-06-04 by dpwe