shorepine/tulipcc #226: Behavior of multiple midi callbacks

dpwe, 2024-06-05:
The guidance for MIDI input functions is to use `tulip.midi_add_callback(f)` to register a function that will be called whenever the MIDI input queue is not empty.  The registered function is supposed to call `tulip.midi_in()` itself, then process the bytes it receives.

It's possible to register multiple functions with further calls to `tulip.midi_add_callback(g)` etc., but these functions are called after the existing registered functions.  Assuming the first callback function called `tulip.midi_in()`, the event that triggered the callbacks has now been consumed, and the subsequent callback functions will get empty returns from `tulip.midi_in()`.  This doesn't seem useful.

We should either (a) allow only one `midi_callback` function, with each new registration removing the previous one, or (b) handle emptying the queue and passing bytes to the callbacks as part of the callback manager.  For instance, we could make the callback signature be `callback_fn(midi_bytes)`, and pass the same midi_bytes array to every registered function when invoking the callbacks.

bwhitman, 2024-06-06:
Good catch. I _want_ the midi_callback to receive the MIDI bytes that triggered the callback, but you can only send a single small integer from C (where we get the MIDI bytes) to Python in Tulip right now. This is due to the way MP calls are scheduled: https://github.com/orgs/micropython/discussions/14164 . In that discussion, you can see that some people work around this for e.g bluetooth packets by having a ring buffer. I haven't tried that yet, but it probably makes sense for Tulip?

The only-allowing a single callback is sad for me because then the `midi.py` callback (that boots when Tulip does) would be the only one, and we couldn't have a separate one for the sliders in juno6 when it boots. 

A new solution I just invented in my head is we do only have a single C -> Python callback, and let the always running `midi.py` receive it, but the `midi.py` callback then delegates out to other Python-registered callbacks. Like we make a `midi_add_callback` in pure Python in `midi.py` instead of worrying about `mp_sched_schedule`. I think that works too.

bwhitman, 2024-06-06:
Check out this proposed fix, works for me: 

https://github.com/bwhitman/tulipcc/pull/227

closed 2024-06-07 by bwhitman