shorepine/tulipcc #231: Tulip Desktop launches a blank screen on Debian

Ekolide, 2024-06-12:
Hello!

Having built Tulip Desktop on Debian 11, it does not launch properly. Instead, it creates a blank screen, my fans starts on full blast, and `htop` reports the process as using 101% CPU. Killing the process from the terminal or killing the window exits the program with "Segmentation fault"

Let me know if I can be of any help!

bwhitman, 2024-06-12:
Hm, i just tried it in Debian 12 on my VM here and it worked as designed 

[image]

If you can't upgrade, can you try running `gdb dev/tulip` and then `run` and then `bt` when it crashes ?

Ekolide, 2024-06-12:
Alright, so it doesn't crash on it's own, but it does provide this error when quitting the non-functioning window.

```
0x00005555556fef55 in display_bounce_empty (bounce_buf=<optimized out>, pos_px=pos_px@entry=0, len_bytes=len_bytes@entry=12288, user_ctx=user_ctx@entry=0x0) at ../shared/display.c:231
231             memcpy(b_ptr, bg_lines[y], H_RES);
```

Not understanding it, I figured maybe it has to do with me using a tiling window manager, so I modified my config to make the Tulip window launches in a floating window. But that just made it worse. Now it flickers between a non-functioning floating window, and a tiled window calling itself "SDL Output", and gdb produces the same error. 

Being that it flickers between these two states, there is some output, but I guess this isn't what you're looking for? It just keeps incrementing the LWP ID.

```
[New Thread 0x7ffff4df3700 (LWP 99836)]
[New Thread 0x7fffeffff700 (LWP 99837)]
[New Thread 0x7fffef7fe700 (LWP 99838)]
[New Thread 0x7fffeeffd700 (LWP 99839)]
[New Thread 0x7fffee7fc700 (LWP 99840)]
[New Thread 0x7fffedffb700 (LWP 99841)]
[New Thread 0x7fffed7fa700 (LWP 99842)]
[New Thread 0x7fffecff9700 (LWP 99843)]
[New Thread 0x7fffed7fa700 (LWP 99842)]
[New Thread 0x7fffecff9700 (LWP 99843)]
[New Thread 0x7fffb3fff700 (LWP 99844)]
[Thread 0x7fffed7fa700 (LWP 99842) exited]
[Thread 0x7fffeeffd700 (LWP 99839) exited]
[Thread 0x7fffef7fe700 (LWP 99838) exited]
[Thread 0x7fffeffff700 (LWP 99837) exited]
[Thread 0x7ffff4df3700 (LWP 99836) exited]
...
[New Thread 0x7ffff4df3700 (LWP 99988)]
[New Thread 0x7fffeffff700 (LWP 99989)]
[New Thread 0x7fffef7fe700 (LWP 99990)]
[New Thread 0x7fffeeffd700 (LWP 99991)]
[Thread 0x7fffeeffd700 (LWP 99991) exited]
[Thread 0x7fffef7fe700 (LWP 99990) exited]
[Thread 0x7fffeffff700 (LWP 99989) exited]
[Thread 0x7ffff4df3700 (LWP 99988) exited]
... etc.
```

bwhitman, 2024-06-14:
I am able to repro this on Debian 11 in a VM. My VM is locked to 1024x768 so it's hard to get the entire TD display on screen, but if H_RES is 1024 / V_RES 600 (the default) it does show a black screen and segfault on quit. (This doesn't happen on my Deban 12 VM!) 

Can you try to edit `tulip/shared/display.h` and change H_RES from 1024 to 800 and V_RES from 600 to 480, rebuild, and see if it still crashes for you? That will give me some clues.

Ekolide, 2024-06-14:
Thank you! The behavior is mostly similar. Now the flickering stops after a short while, but the window still does not properly start, and CPU still kicks to 100%.

So here's the gdb output when I quit the program, followed by a backtrace.

```
Thread 1 "tulip" received signal SIGSEGV, Segmentation fault.
0x00005555556fef60 in display_bounce_empty (bounce_buf=<optimized out>, pos_px=pos_px@entry=0, len_bytes=len_bytes@entry=7200, user_ctx=user_ctx@entry=0x0) at ../shared/display.c:231
231             memcpy(b_ptr, bg_lines[y], H_RES); 
(gdb) bt
#0  0x00005555556fef60 in display_bounce_empty (bounce_buf=<optimized out>, pos_px=pos_px@entry=0, len_bytes=len_bytes@entry=7200, user_ctx=user_ctx@entry=0x0) at ../shared/display.c:231
#1  0x00005555556a8b57 in unix_display_draw () at ../shared/desktop/unix_display.c:218
#2  0x00005555556a80f6 in main (argc=1, argv=0x7fffffffe148) at main.c:909

```

Upgrading is an option, but if this roots out some underlying issue that might cause issues down the line then maybe it's worthwhile pursuing? I'm not knowledgeable enough to make out the nuances of it, though.

bwhitman, 2024-06-14:
Hm, ok, i'll keep poking at it here. I am not certain why TD works fine on macOS/Fedora/Ubuntu but not Debian (we had a similar crash in Arch recently as well). I think it's something to do with SDL. I'll try to get it stable on my Debian 11 machine and then ask you to try again once I do.

bwhitman, 2024-06-16:
I think I might have fixed this! I moved us off the SDL_SoftwareRenderer and to its default renderer. This now works for me on Debian 11!

[image]

Give it a go and let me know.

Ekolide, 2024-06-17:
Hi! Thank you very much, there's progress on my machine as well. It now displays properly, but I can't interact with most anything. The initial status message doesn't show up. Using CTRL-D crashes the program. Using `edit` doesn't start the editor, but rather puts me into a prompt. I managed to use `newfile` to create a file, even though it took a while for it to understand when I was using CTRL-D to exit, at which point it crashed. But my test file shows up using `ls`, at least. History is working, I can use arrows to go through previous commands, and use CTRL-I and TAB. And `cat` properly displays the contents of the file I made using `newfile`.

[image]

Ekolide, 2024-06-17:
Oh what you see on the image is exactly what happens. I enter `edit("test")` and it puts me back at the prompt. Except no keyboard input works, so I CTRL-D, putting the cursor on the next line but hanging the system. No GUI, no status bar at the bottom.

dromer, 2024-08-03:
For me on Debian 12 it segfaults immediately:

```gdb
(gdb) run
Starting program: /home/dreamer/Sources/_audio/tulipcc/tulip/linux/dev/tulip 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff06ba6c0 (LWP 289679)]
[New Thread 0x7fffefeb96c0 (LWP 289680)]
[New Thread 0x7fffef6b86c0 (LWP 289681)]
[New Thread 0x7fffee6b66c0 (LWP 289682)]
[New Thread 0x7fffedeb56c0 (LWP 289683)]
[Thread 0x7fffee6b66c0 (LWP 289682) exited]

Thread 1 "tulip" received signal SIGSEGV, Segmentation fault.
0x00007ffff4a5d0cc in ?? () from /lib/x86_64-linux-gnu/libnvidia-glcore.so.535.183.01
(gdb) bt
#0  0x00007ffff4a5d0cc in ?? () from /lib/x86_64-linux-gnu/libnvidia-glcore.so.535.183.01
#1  0x00007ffff4a5dc19 in ?? () from /lib/x86_64-linux-gnu/libnvidia-glcore.so.535.183.01
#2  0x00007ffff4a68673 in ?? () from /lib/x86_64-linux-gnu/libnvidia-glcore.so.535.183.01
#3  0x00007ffff46f41a8 in ?? () from /lib/x86_64-linux-gnu/libnvidia-glcore.so.535.183.01
#4  0x00007ffff7e1807a in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#5  0x00007ffff7e0b7a1 in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#6  0x00007ffff7e1426d in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#7  0x00005555556a64cd in unix_display_draw () at ../shared/desktop/unix_display.c:203
#8  0x00005555556a5952 in main (argc=1, argv=<optimized out>) at main.c:909
```

bwhitman, 2024-08-03:
@dromer I just tried on my Debian 12 VM (arm64) and it runs fine. I made #295 to try to find us some help!