shorepine/tulipcc #253: Question: Will MIDI be implemented in the Tulip desktop?

rgbeitor, 2024-07-11:
Hello.

First of all thank you for your awesome work with Tulip.

This is more a question than an issue anyhow:

Will there be MIDI in Tulip desktop?

I'm interested in a standalone unit (hardware) to program some generative sequencers (euclidean, etc), but I wanted to check it beforehand (if possble).

Thanks in advance for the answer.

BR

bwhitman, 2024-07-11:
MIDI is available on the MacOS version of Tulip Desktop. It should just work using the "IAC" bus using other MIDI software on your computer.

MIDI is not yet available on Tulip Desktop for Linux or Windows. I'm hoping someone who knows how to do MIDI on Linux can contribute a patch.

rgbeitor, 2024-07-12:
Thank you for your quick response.

Yeap, I'm using linux. I was building it in an Ubuntu Studio.

MIDI linux should be similar to MaCOS (in the background MaCOS is also UNIX based), but yes, probably for an specialist on it.