shorepine/tulipcc #259: Running Tulip on Raspberry Pi

pravin-vaz, 2024-07-20:
Hi team, thanks for this fantastic piece of music software. 
Currently running a Pi4, with a pisound card and their custom raspbian image(blokas)
Was keen to know if I can run Tulip on it? 
It seems to install both the packages when I used the Ubuntu cmds, but can't seem to run the cd tulip/linux command (error: no such file or directory).
Again, I am noob and probably doing something wrong. 
Looking forward for some support thanks.

bwhitman, 2024-07-20:
Yes it will totally work on a pi4. 

Did you checkout the repository ? You should see a tulipcc folder. If so, first cd into that, then cd tulip/linux, then run ./build.sh and then dev/tulip. Let us know how that goes !

bwhitman, 2024-07-29:
Closing for now. If you're having problems, please file another issue with any errors or problems you're seeing.

closed 2024-07-29 by bwhitman