shorepine/tulipcc #295: Linux Tulip Desktop megaissue, we need help with maintaining

bwhitman, 2024-08-03:
We're having various problems with Tulip Desktop in Linux. There are probably a few unrelated issues #231 #230 #121 

I personally don't use Linux regularly, and it shows. When I test it, I run Tulip Desktop just fine in VMWare on my Mac on Ubuntu 22 and Fedora 39 in VMs. I run the ARM linux versions to test. I don't have easy access to an intel machine, or one with a real GPU, or etc

I would **love** for someone who's more linux-savvy to help me get this app into shape so we stop having so many problems. Most of them seem to be to do with SDL. SDL works for me on Mac and the linux-es I test on, however.

There's also #253, which is a feature we need to add in for MIDI. I don't know enough about Linux MIDI to pull this off.

Can anyone help in the short term? Longer term I do really think we should just move Tulip Desktop to the web (#182) and just maintain one platform, but that'll be a while. In the interim I would love to have someone help us out getting better at running Desktop in linux!

bwhitman, 2024-08-03:
Logistics note: we do _build_ the linux version as a commit hook in GitHub here, so we are tracking that it always builds fine. Maybe there's a way to also make a run hook to test the app starts in the GitHub task runner?

znmeb, 2024-08-03:
I don't know anything about GitHub Actions, etc., but I can certainly test and troubleshoot on various Linux environments. I have an Alienware Aurora with an NVIDIA RTX 3090 on Windows 11 / WSL, which can run Ubuntu, Debian, Arch and Fedora. NVIDIA only supports the GPU in Ubuntu, however. 

If you're interested in "native" Windows, Tulip will probably build and run in a [Mambaforge](https://mamba.readthedocs.io/en/latest/index.html) virtual environment. This is a Python `conda` based package and environment manager that supports just about everything Python and GPU, including PyTorch and TensorFlow. It also has many common libraries and build tools like `git`, `gcc`, `make`, `CMake` and `ninja`. It runs on 64-bit Windows, MacOS and Linux.

I have an Acer Nitro 5 with an NVIDIA GTX 1650 on Universal Blue Bluefin 40 (a superset of Fedora Silverblue 40). Bluefin/Silverblue is a Fedora Atomic Desktop designed for "container-native" workflows. The host has a GNOME desktop and can run Fedora apps, but most of the work is done in either `Distrobox` or Docker containers. `Distrobox` containers can be just about any Linux distro that has a Docker image on Docker Hub, but I only use Fedora 40, Ubuntu "jammy" and "noble" and Debian "bullseye" and "bookworm" regularly. I occasionally use Arch.

I also have a few Raspberry Pi Zero 2 Ws. They can run either Ubuntu "jammy" or Debian "bullseye" 32 or 64 bits, but they only have 512 MB of RAM, which isn't enough to run Ubuntu "noble" or Debian "bookworm". 512 MB of RAM isn't enough for a desktop but they should run the Tulip desktop assuming the compiles can run in 512 MB of RAM.

murilopolese, 2025-01-04:
I have just built the project for linux and got this segmentation fault error trying to run it:
```
$ gdb dev/tulip
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from dev/tulip...
(gdb) run
Starting program: /home/murilopolese/Projects/tulipcc/tulip/linux/dev/tulip 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff0200640 (LWP 154215)]
[New Thread 0x7fffef800640 (LWP 154216)]
[New Thread 0x7fffeee00640 (LWP 154217)]
[New Thread 0x7fffeda00640 (LWP 154218)]
[New Thread 0x7fffed000640 (LWP 154219)]
[Thread 0x7fffeda00640 (LWP 154218) exited]
MicroPython v1.24.0-preview.409.g82e69df33 on 2025-01-04; Tulip4 with linux
>>> 
Thread 1 "tulip" received signal SIGSEGV, Segmentation fault.
0x00007ffff465d0cc in ?? () from /lib/x86_64-linux-gnu/libnvidia-glcore.so.535.183.01
```
I'm excited about the web idea but definitely would also love to have it running standalone on linux! :heart_eyes:

jobf, 2025-01-04:
I've got the same thing on a desktop with nvidia graphics card, however it works perfectly on a laptop with intel integrated graphics running the same linux.

I don't know what the fix is but I'm wondering if something that needs enabling for SDL to work with the nvidia (because nvidia and linux is too fun).