shorepine/tulipcc #309: Tulip headless firmware builds

bwhitman, 2024-08-09:
We've had some chat about a "headless" Tulip: #287 #115

And working with people trying AMY on Arduino on the discord, i'm realizing how complex it is to do things like have a MIDI input control AMY in C, especially patches, voices, parameters. And we've already done a lot of work in Tulip covering up that complexity in Python, with our `midi.py` / `voices.py` etc polyphonic / multimbral `Synth` units. 

So why not start building a version of Tulip without the display core, and somehow run-time addressable pins for I2S, I2C, MIDI? And the board boots into a UART with a Python REPL, just plain Micropython, but with the Tulip modules baked in and AMY running. That way, people could even port their Tulip creations to a headless synth, even an Alles board. Or a RP2040, or a RP2350, or a Teensy, or a Daisy, or etc etc. 

This is in research stage, we'll have to figure out a lot to get this really going, but I'd love any feedback others have on this idea, and especially boards you'd like to be supported.

znmeb, 2024-08-09:
Boards: 
- Raspberry Pi Pico for sure, including the new RP2350 boards. Anything with an RP2040 should work.
- ESP32-S3 and ESP32-C6
- Rebel Technology OWL 2 and 3
- Electro-Smith Daisy
- Teensy 4.1

There are quite a few Eurorack modules with a USB port, a micro-controller and knobs / faders / analog I/O, etc. Allen Synthesis EuroPi, Electro-Smith Daisy Patch and patch.init(), Befaco Lich. Rebel Technology OWL 2 and OWL 3 devices exist in desktop, Eurorack and "stompbox" hardware configurations.

Connectivity options:
- USB MIDI and audio
- TRS MIDI
- Analog audio and control voltage
- USB and UART serial

dpwe, 2025-02-04:
Yes, how long until the ESP32-P4 is available for this kind of thing?

bwhitman, 2025-02-04:
This in-design AMYboard will be the same chip as Tulip -- the ESP32-S3. However, it should support more polyphony and RAM than Tulip as it doesn't need to run all the Tulip things. If you're bound by something you don't think an ESP32-S3 @ 240Mhz dual core + 8MB can do, let us know so we can help figure it out. We're able to squeeze quite a lot of synth / effects power out of this chip and i don't currently see "horesepower" as a blocker for us. We're trying to balance cost with power for people that want to build 100s of synths/effects units out of AMYboards. 

I'm excited about the P4, it clocked roughly double the polyphony as the S3 in early tests. But it's not yet available for manufacturers for more than a few parts for testing. It doesn't have any WiFi on package like the S3, so would require a separate C6 or similar ESP chip as well. The bare P4 chip looks like it's also going to be about double the cost of a S3 at the same quantities. The tooling is also quite in flux with a lot of scary errata still. I would imagine a future AMYboard or Tulip in 2026 could have one, maybe? 

Re: other chips -- something like a cortex M7 with external RAM could be great with AMY, but will increase the cost considerably -- the [very similar Daisy Pod/Seed is $71](https://electro-smith.com/products/pod) -- and would require a lot of new board design / tooling work on our end. The RP2530 is nice but really is about the same clocks/capacity as the ESP32S3 for AMY and would need external RAM and WiFi as well. 

That's "our" current thoughts on the MCUs for synth scene in early 2025. If anyone's got something I missed, let's chat about it here.

wjbuys, 2025-08-01:
Hi! Is there a working headless build available yet? I've tried building the `tulip/amyboard` IDF subproject, but the compile fails for me with a bunch of these errors:

```
../py/emitinlinethumb.c:153:9: error: initializer-string for array of ‘unsigned char’ truncates NUL terminator but destination lacks ‘nonstring’ attribute (4 chars into 3 available) [-Werror=unterminated-string-
initialization]
  153 |     {0, "r0\0"},
```

The amyboard build looks like exactly what I need (even uses the same ram/rom configuration), and would be great if I could get this working, happy to help debug!

wjbuys, 2025-08-01:
(never mind, that's a micropython/GCC15 issue that's already fixed upstream: https://github.com/micropython/micropython/pull/17269)