shorepine/tulipcc #327: Tulip Desktop is a web page, not a native app

bwhitman, 2024-08-27:
We want to simplify development and ease of use for Tulip Desktop, our Tulip simulator. 

Tulip Desktop is crucial for Tulip - it helps people learn about Tulip and also helps them write programs for Tulip. 

It is currently based on the `unix` port of Micropython and SDL, to simulate the screen. AMY works as is. We (Brian and Dan) do most of our work on macOS, and the linux/WSL ports have various problems that we can't fix. The linux/WSL ports also have to compiled from source. 

We want to build Tulip Desktop for the web, using Micropython's `webassembly` port, AMY's web port, and `emscripten`'s SDL support (or find another frame buffer-like canvas object.) This will let anyone just go to (e.g.) https://tulip.computer/desktop and have their own file system and be able to interact with Tulip on the web with no software install.

znmeb, 2024-08-27:
Why not Web Audio and Web MIDI? Do you really need an Emscripten SDL2??

bwhitman, 2024-08-27:
SDL is not for audio and MIDI (we will use native Web Audio (Worklets) and MIDI) but for the framebuffer, keyboard and touchscreen/"mouse".

znmeb, 2024-08-27:
Ah ... I don't know much about the video stuff.

bwhitman, 2024-11-23:
This work is well underway, see latest here: https://tulip.computer/run/

znmeb, 2024-11-23:
Looks great! I'm working with AMY in Arduino on Pico Ws at the moment, but my project will definitely need some desktop components to supply data to the Pico version.

bwhitman, 2025-02-04:
Calling this done! We launched Tulip Web at https://tulip.computer/run/ -- you can share code, use Tulip World, there's an editor, etc.

closed 2025-02-04 by bwhitman