shorepine/tulipcc #336: PROPOSAL: Tulip could have an SD card to load samples and code from

bwhitman, 2024-08-29:
An optional microSD card is a simple add for Tulip, assuming we have enough pins for it and it doesn't conflict with other SPI operations. It could mount a `/sd` filesystem alongside the built in one. Software could stream samples over "disk" and it could be a way to transfer code / files between a desktop and Tulip.

dakotawinslow, 2025-08-06:
I think this is a good idea! It could match Lilygo's implementation on the T-Deck for ease of maintenance

mirkokral, 2026-09-16:
> I think this is a good idea! It could match Lilygo's implementation on the T-Deck for ease of maintenance

iirc, lilygo's implementation is sharing an SPI bus with the display with CS. This is not a good way to do it, and if possible, it's best to keep the display bus only to the display.