shorepine/tulipcc #344: Tulip needs to understand MIDI "running status"

dpwe, 2024-08-31:
According to the MIDI description at http://midi.teragonaudio.com/tech/midispec/run.htm,
> The MIDI spec allows for a MIDI message to be sent without its Status byte (ie, just its data bytes are sent) as long as the previous, transmitted message had the same Status. This is referred to as **running status**.

This could explain the various reports of missing note-on/note-offs we've seen when receiving input from external MIDI gear.  This is specifically to reduce bandwidth and latency on serial-line MIDI, so it may be selectively employed with MIDI hardware but not pure software busses.

The effect is that "extra" data bytes without a Status byte (i.e., a byte with its top bit set) are implicitly assumed to have the same Status byte as the preceding command.  So a sequence like:
```
0x90  # Note-on Status, channel 0
0x3c  # Note 60
0x7f  # Velocity 127
0x90  # Note-on Status, channel 0
0x40  # Note 64
0x7f  # Velocity 127
0x90  # Note-on Status, channel 0
0x43  # Note 67
0x7f  # Velocity 127
```
can be sent equivalently without the repeated note-on Status bytes as:
```
0x90  # Note-on Status, channel 0
0x3c  # Note 60
0x7f  # Velocity 127
0x40  # Note 64
0x7f  # Velocity 127
0x43  # Note 67
0x7f  # Velocity 127
```

Currently, the Tulip midi parser doesn't understand this.

Many thanks to @Speccary for tracking this down!

Speccery, 2024-08-31:
Thanks for looking into this, I'll be looking forward to a release with the fix - or taking a look at the code myself.

bwhitman, 2024-09-06:
@Speccery please let us know if the latest Tulip (use `tulip.upgrade()` or Tulip Desktop build) works for you with your setup. We believe we've implemented support for running status and sense bytes.

AntoineDoinel69, 2024-09-13:
@bwhitman
I recently built a DIN to jack cable, and tried before and after the last update: it seems to work.
There are still problems intermixing notes with CC messages, but I'm not sure if its related.

bwhitman, 2024-09-19:
Hey @Speccery I've got my SH-4D wired up to Tulip over MIDI DIN, and sync is on MIDI and the sequences are playing in sync and with no dropped notes. Do you have a reduced test case we can try to show your error?

closed 2024-10-14 by bwhitman