shorepine/tulipcc #356: amy.stop() returns ImportError

jack2702, 2024-09-07:
Error retruned:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "amy.py", line 306, in stop
ImportError: no module named 'libamy'

bwhitman, 2024-09-07:
Hi, can you give more information? What were you doing and what did you expect to happen? Have you installed the AMY python module first?

bwhitman, 2024-09-07:
I just realized you've posted this in the Tulip issues list. Are you using a Tulip? If so, you don't need to ever call `amy.stop()`. It's always running on a Tulip. `amy.reset()` may be what you want.

jack2702, 2024-09-07:
> I just realized you've posted this in the Tulip issues list. Are you using a Tulip? If so, you don't need to ever call `amy.stop()`. It's always running on a Tulip. `amy.reset()` may be what you want.
> 

Oh that makes sense. Thanks

closed 2024-09-07 by bwhitman