shorepine/tulipcc #367: Unable to connect to WiFi

steveh, 2024-09-09:
Running on a Tulip CC with unmodified `tulip-firmware-TULIP4_R11.bin`.

Attempting to connect to a random neighbours network (with the wrong password) returns an empty prompt after a second or two.

```
>>> tulip.wifi('neighbours-wifi', 'wrong-password')
>>>
```

Connecting to my wifi, with the right password or the wrong password, crashes the system.

```
>>> tulip.wifi('my-wifi', 'right-password')
E (7753) intr_alloc: No free interrupt inputs for AES interrupt (flags 0xE)
E (7754) esp-aes: Failed to allocate AES interrupt 261

abort() was called at PC 0x4210da9c on core 0

Backtrace: 0x4037599e:0x3fcddec0 0x40383f59:0x3fcddee0 0x4038b242:0x3fcddf00 0x4210da9c:0x3fcddf70 0x4210d449:0x3fcddfa0 0x42103a21:0x3fcddfc0 0x4213cf0a:0x3fcddfe0 0x42140fae:0x3fcde1f0 0x421428d0:0x3fcde210 0x4213c2c5:0x3fcde5f0 0x4212434f:0x3fcde660 0x421253de:0x3fcde6b0 0x421294df:0x3fcde6d0 0x421260dd:0x3fcde750 0x4212ceb4:0x3fcde770 0x4212d2ba:0x3fcde7c0 0x4212e181:0x3fcde810 0x4212e1fc:0x3fcde890 0x42127938:0x3fcde8e0 0x42127d8b:0x3fcde930 0x4212c77b:0x3fcde960 0x4212c79a:0x3fcde980 0x4212c310:0x3fcde9a0 0x4212c593:0x3fcde9c0 0x4216d185:0x3fcde9e0

ELF file SHA256: 74ff440f3

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376160
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0xf08
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbb0
load:0x403cc700,len:0x2acc
entry 0x403c98a0
Starting MIDI on core 0
UART MIDI running on core 0
Starting USB host on core 1
Starting display on core 0
Starting touchscreen on core 0
Resetting touch i2c RST pin twice
this is the TULIP SPECIAL esp_lcd
Starting Alles on core 1
i2s started blck 8 dout 5 lrck 2
StaStarting Sequencer (timer)

err for writing to 0x5D is 0 ESP_OK
MicroPython v1.20.0-244-g2ede7c6c2 on 2024-09-07; TulipCC with ESP32S3
Type "help()" for more information.
>>>
```

My guess is that something about my WiFi is not implemented in TulipCC. WPA3 perhaps. I worked around it by resetting my network settings back to their automatic values. FWIW it works with an ESP8266 on ESPHome.

---

For reference, here's how my Unifi UDM-SE and AP was configured:

[image]

bwhitman, 2024-09-10:
Hi! I checked out these errors and i'm afraid i don't know how i can help just yet. I haven't seen `Failed to allocate AES interrupt` reported anywhere else online and am not sure where it's coming from. We do support WPA3, I believe (it's turned on in the chip support settings). I'd love to help figure this out, so will try to investigate a bit more in the `esp-idf` source to see what that error can mean.

steveh, 2024-09-10:
I think it's coming from here: https://github.com/espressif/esp-idf/blob/d3be7bda056dfcb7edbac4cf8757ec1ac7d0a1ea/components/mbedtls/port/aes/dma/esp_aes.c#L187

But it looks like there's been a lot of activity in that repo since, that line was rewritten in https://github.com/espressif/esp-idf/commit/2abb656ba2d14720ca7c22c636244cd9351b4643#diff-83729b211ef1fb0aad2272032878f600a9990a2288f9d83f470141990db88bdc

bwhitman, 2024-09-10:
OK, I think the next step is to wait for #315 , and hope they've made some bug fixes to support your wifi setup .

michael-hatz, 2024-09-16:
I ran into the same problem and played around a bit - WPA3 and WPA2/WPA3 networks don't seem to work. Trying to connect to them with a correct password will crash the device. Changed to WPA2 and it did work without problems.

bwhitman, 2024-09-16:
OK, thanks for the update. Let's hope that the later ESP-IDFs fix this issue, we'll be upgrading those soon.

bwhitman, 2024-10-12:
We think we've added WPA3 support in the latest Tulip firmware. If you can connect to something to get it to do `tulip.upgrade()`, or flash it manually, let me know how it goes!

closed 2024-10-29 by bwhitman