shorepine/tulipcc #378: PROPOSAL: Tulip could get SYSEX messages passed to the MIDI callback

bwhitman, 2024-09-11:


jobf, 2025-01-04:
I'm guessing that there is a reason for not passing the SYSEX through but what would be needed to get this working and are there any drawbacks to doing so?

Most of the tools I am looking to build with tulip involve bidirectional SYSEX via MIDI or USB so I'm keen to know more.

dpwe, 2025-01-04:
I don't think there's any particular problem with passing through sysex.  I think no-one has needed it yet (badly enough to add it).  I don't have enough MIDI setup to actually test it, much less need it.

It looks like it's just a matter of adding some code to `tulip/shared/midi.c:convert_midi_bytes_to_messages()` to echo sysex bytes to `midi_out()` rather than just ignoring them.

bwhitman, 2025-01-04:
Yes, it would be relatively simple to add, but since Dan nor I do much with SYSEX we can't really test it and make sure the limits we've set work for most people. let us know the use case, and if you can, the size of the SYSEX packets your software would want to work with. 

As I understand it, completing this task means `tulip.midi_in()` can also receive sysex messages of arbitrary length (including the `F0` and `F7` markers) up to (say) 255 bytes. To do this, we use the existing `convert_midi_bytes_to_messages()` code to keep a running buffer of sysex messages (we parse these but do not save them anywhere). We also need to increase the midi queue to handle larger messages (we have a max message length of 3). Then we need to do the same for the midi output for USB (midi TRS output should just work for sysex today), and also confirm that midi input for USB works (it should, but i'm nervous about the 4 byte packets, we haven't tested this.)

jobf, 2025-02-26:
The use case varies, in some cases it is for updating patch parameters on older synths. In other cases communicating with USB MIDI controllers to set some config or send data to be displayed on an OLED screen for example.

I'm not sure on the packet count, unless you mean the message length. The longest message I'm looking to send is 1881 bytes, which seems excessive now that I've written it down but may need sending fairly frequently.

bwhitman, 2025-03-25:
We are closer on this -- we now support MIDI SYSEX in and out. But for SYSEX in, we do not yet pass this onwards to Python. You can definitely send SYSEX out (either USB MIDI or UART MIDI) from Tulip Python, and hopefully soon we'll have a way to parse it in Tulip Python when it comes in from other devices.

bwhitman, 2025-03-26:
This is done! Please check out the docs on how to access incoming SYSEX data in Tulip.

closed 2025-03-26 by bwhitman