shorepine/tulipcc #399: Midi not working

zaxbatch, 2024-10-10:
Midi isn't working on my device.  I'm using a keystep 32. Midi out from the keystep to midi in to the tulip. Any help is appreciated.

bwhitman, 2024-10-10:
Tulip should play Juno-6 patch 0 on boot from any MIDI in on channel 1. Some steps to debug:

 - Does the MIDI out from the keystep work on other MIDI devices?
 - Are you certain the MIDI out from the Keystep is playing on the first MIDI channel? 
 - Are you using a MIDI->TRS **type a** converter? It has to be Type A. 
 - Does the Tulip make other sounds, like if you `run('voices')` and hit the on-screen keyboard you hear a Juno-6 patch?

zaxbatch, 2024-10-11:
I think my midi cable is type b. It came with an arturia product, and I
learned that's what they use. I ordered type a cables today, and I also got
usb midi to work. I'm optimistic about the new midi cables working. They
should be here in 5 days. I'll keep you posted. Thanks for your help.

zaxbatch, 2024-10-17:
The type a midi cables solved the problem. Thanks for your help.

closed 2024-10-17 by zaxbatch