shorepine/tulipcc #402: Keyboard not recognised after update

mike-kelly, 2024-10-12:
I plugged in my keyboard to Tulip, and was able to enter commands. After setting up wifi, and running update, the Tulip rebooted but the keyboard is no longer recognised. Nothing happens at the command line.
The version displayed on screen is MicroPython v1.24.0-preview.409.g82e69df33.dirty.

I am using a usb A to usb C adaptor for the keyboard to Tulip connection, but this worked OK before running the update. The keyboard doesn't have a hub.

bwhitman, 2024-10-12:
OK, thanks for the report! Can you tell me what keyboard you are using? And it would be very helpful if you could show me the serial debug output of tulip while it boots -- it shows a lot of USB information.

mike-kelly, 2024-10-12:
Hi! The keyboard is this one: https://splitkb.com/collections/keyboard-kits/products/aurora-sofle-v2

mike-kelly, 2024-10-12:
Boot output without keyboard connected:
```
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0xf08
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbb0
load:0x403cc700,len:0x2acc
entry 0x403c98a0
Starting MIDI on core 0
UART MIDI running on core 0
Starting USB host on core 1
usb_host_install: 0x0
usb_host_client_register: 0x0
Starting display on core 0
Starting touchscreen on core 0 
Starting Alles on core 1
i2s started blck 8 dout 5 lrck 2
StaStarting Sequencer (timer)
MicroPython v1.24.0-preview.409.g82e69df33.dirty on 2024-10-12; TulipCC with ESP32S3
Type "help()" for more information.
>>> Resetting touch i2c RST pin twice
touchscreen OK at 0x5D
```

Tail of output when booting with Sofle keyboard connected:
```
** INTERFACE_DESC(size=9)
** bInterfaceNumber: 1
** bAlternateSetting: 0
** bNumEndpoints: 2
** bInterfaceClass: 0x03
** bInterfaceSubClass: 0x00
** bInterfaceProtocol: 0x00
** iInterface: 0
We are using a different type of mouse claim
config_desc->val[43]: type=33 length=9
config_desc->val[52]: type=5 length=7
*** ENDPOINT_DESC(size=7)
*** bEndpointAddress: 0x82 (In)
*** bmAttributes: 0x03 (Interrupt)
*** wMaxPacketSize: 32
*** bInterval: 1
Setting mouse or 0x03/0/0 to 32 from MPS 32
Redirecting this endpoint to KB (0x03/0/0)
Setting KB to 32 from MPS 32
USB boot keyboard ready
config_desc->val[59]: type=5 length=7
*** ENDPOINT_DESC(size=7)
*** bEndpointAddress: 0x03 (Out)
*** bmAttributes: 0x03 (Interrupt)
*** wMaxPacketSize: 32
*** bInterval: 1
config_desc->val[66]: type=4 length=9
** INTERFACE_DESC(size=9)
** bInterfaceNumber: 2
** bAlternateSetting: 0
** bNumEndpoints: 1
** bInterfaceClass: 0x03
** bInterfaceSubClass: 0x00
** bInterfaceProtocol: 0x00
** iInterface: 0
We are using a different type of mouse claim
config_desc->val[75]: type=33 length=9
config_desc->val[84]: type=5 length=7
*** ENDPOINT_DESC(size=7)
*** bEndpointAddress: 0x84 (In)
*** bmAttributes: 0x03 (Interrupt)
*** wMaxPacketSize: 32
*** bInterval: 1
Setting mouse or 0x03/0/0 to 32 from MPS 32
Redirecting this endpoint to KB (0x03/0/0)
Setting KB to 32 from MPS 32
USB boot keyboard ready
StaStarting Sequencer (timer)
MicroPython v1.24.0-preview.409.g82e69df33.dirty on 2024-10-12; TulipCC with ESP32S3
Type "help()" for more information.
>>> Resetting touch i2c RST pin twice
touchscreen OK at 0x5D
```

bwhitman, 2024-10-12:
Really helpful! I think I see an issue. I'll test this locally and let you know here when I have a fix for you to try!

bwhitman, 2024-10-13:
When you get a chance, can you try `tulip.upgrade()` again? (You can use the serial terminal to type since your keyboard doesn't work!) I made a change that I think will help your setup.

mike-kelly, 2024-10-13:
That worked, thank you!

closed 2024-10-13 by mike-kelly