shorepine/tulipcc #403: The Mouse movement is EXTREMELY sensitive, and doesn't seem to let me click anything

thelegendofmario, 2024-10-12:
When I connect a mouse using a usb to usb-c adapter, the mouse is very sensitive to movement up and down, and if I try to click on the launcher, It doesn't work.

thelegendofmario, 2024-10-12:
also, the mouse icon appears if I plug in a wired usb keyboard with the same usb to usb-c adapter

bwhitman, 2024-10-12:
I figure we'll eventually want to set mouse speed/acceleration, and maybe certain mice report position differently. Which mouse are you using? 

Regarding the second issue, is this happening after a reboot without the mouse ever plugged in?

thelegendofmario, 2024-10-14:
I'm using some old logitech mouse, and I tried plugging in the mouse and then turning it on, and it still won't click.

bwhitman, 2024-10-29:
Hi, i'd love to help more, but need more information. Can you get access to the Tulip boot messages? https://github.com/shorepine/tulipcc/blob/main/docs/troubleshooting.md#see-the-diagnostic-output-of-a-tulip

thelegendofmario, 2025-02-11:
this isn't going anywhere lol I thought I'd help with the clutter of issues

closed 2025-02-11 by thelegendofmario