shorepine/tulipcc #410: USB hubs with a qwerty keyboard and midi keyboard depends on port order

rhooper, 2024-10-22:
When using a USB2 (or 3) hub, powered or not, the port order appears to matter when initializing the usb host.
When a keyboard and midi are both plugged in, only one of Midi or Keyboard will work.

[hub-kbd-midi-swap-midiworks.txt](https://github.com/user-attachments/files/17478529/hub-kbd-midi-swap-midiworks.txt)
[hub-kbd-midi-kbdworks.txt](https://github.com/user-attachments/files/17478530/hub-kbd-midi-kbdworks.txt)
[hub.txt](https://github.com/user-attachments/files/17478531/hub.txt)
[direct.txt](https://github.com/user-attachments/files/17478532/direct.txt)

bwhitman, 2024-10-29:
Thanks for this! I'll investigate. I do see some places we can improve and I'll let you know when you can test a new build.

CelloGit, 2024-12-28:
Great that a USB hub is now supported. It allows for using a keyboard with a 2.4GHz dongle such as the Logitech K400+.
To my surprise even the integrated touchpad triggers the mouse function on the tulipcc but only horizontal movement, needs two fingers for vertical (and very slow as well) and no mouse clicks. Even more surprising since I couldn't get the tulipcc to recognize any other mouse.

Unfortunately, as @rhooper posts above, it seems not (yet) possible to use two USB devices on a hub. Even not toggle them by (dis)connecting. It looks like the tulipcc assigns 1 USB device at  boot and that is it. Reboot if you want to use another device.