shorepine/tulipcc #445: https://tulip.computer/run/ does not work in Firefox private brosing mode

matt-har-vey, 2025-01-10:
### Steps to reproduce

1. Create a new Firefox profile, to assure default settings.
2. Open a private browsing window.
3. Enter https://tulip.computer/run/ in the address bar.
4. Open WebDeveloper tools.

### Result

`SharedArrayBuffer is not defined` visible in JavaScript console.
`ImportError` visible in MicroPython console.

### Details

This also happens in a non-fresh Firefox profile where the "Always use private browing mode" setting is enabled.

Considering HTTP response headers on a cache-skipping reload of the above URL, in the non-private window

```
cross-origin-embedder-policy: require-corp
cross-origin-opener-policy: same-origin
cross-origin-resource-policy: cross-origin
```

are present. In the private window they are absent. I don't know the cause of the difference, or whether it is even feasible to set these policies in a private Firefox window.

bwhitman, 2025-01-10:
Appreciate the report! I do want Tulip Desktop Web and all the other things to run everywhere but private browsing on many browsers will intentionally prevent some of the technologies we're using to make that happen. (We're just caught in the crossfire here -- it's not that we ever try to send data out of tulip or track anyone.) It _looks_ like the thing breaking here is our use of [`mini-coi`](https://github.com/WebReflection/mini-coi) to allow SharedArrayBuffers to work on Github Pages. A path forward is to hose these types of things on a server we can control the headers of.

bwhitman, 2025-01-10:
https://github.com/WebReflection/mini-coi/issues/5

matt-har-vey, 2025-01-20:
Thanks for looking into it so quickly.

Since it is really a Firefox problem https://bugzilla.mozilla.org/show_bug.cgi?id=1320796

and since there's recent activity on that bug on the Mozilla side, I'll be content using Chromium incognito as a workaround until they fix it. I'll leave it to you whether to close the issue. Closed or not, it will exist for others who might run into this.

> it's not that we ever try to send data out of tulip or track anyone

I hope I didn't imply that. I didn't mean to. I've used private browsing for everything for years now, and I forgot it sometimes breaks things.

bwhitman, 2025-01-22:
Oh, no worries! I'll keep an eye on the Firefox issue tracker until this can get resolved, thanks!

closed 2025-01-22 by bwhitman