shorepine/tulipcc #517: tulip web: run UIScreen apps in code editor

bwhitman, 2025-02-13:
We just eval the code editor. It should look for a `def run()`? and `run` it instead. Or something. But for now, if you paste a UIScreen app in the editor and click the green button it doesn't do anything unless you add a `if __name__ == '__main__':` thing. 
See https://share.tulip.computer/gEP7Mu

bwhitman, 2025-02-14:
An annoyance here (self-owned) is that `def run(screen):` overwrites the imported `from tulip import run` in the namespace when running this type of code. They are two different `run`s with the same name.

bwhitman, 2025-02-14:
Is that their style guide for python interpreters that are also the CLI for the entire OS? They think of everything . 

I really want to keep things like bare `ls` and `run` at the REPL. It's a better fix to rename the `run` out of UIScreen apps and/or have the web interpreter be somewhat intelligent about what kind of code it's running.

dpwe, 2025-02-14:
I agree with making `run` resolve to `tulip.run` in the REPL.

What about using `main()` for the main function of the app you're defining in the editor ?

FTR:  I think the bare `ls` etc. are unhappy compromises, because they attempt to bend the syntax of the REPL.  I frequently find myself tricked into thinking I'm typing at a terminal shell, then very rapidly am brought back to reality.  My preference is to embrace the REPL syntax.  Not for nothing was `print variable` was removed in Python 3.x.

[image]