shorepine/tulipcc #568: DX7 Marimba patch sometimes clicks

drepetto, 2025-04-04:
Intermittent clicking on attack when playing the DX7 Marimba patch (and maybe others?) Probably caused by an oscillator that's not being reset at note on. 

REPRO:

midi.config.reset()
s = synth.PatchSynth(patch_number=149)
for t in range(20):
    s.note_on(note=36, velocity=1, time=300*t)
    s.note_off(note=36, time=300*t+300)

drepetto, 2025-04-04:
To me, this makes the errant clicks very clear. If it were part of the attack I would expect it to be the same for each beat. Maybe I'm just pushing the patch beyond what it's meant to do, I don't know. But 36 isn't an egregiously low MIDI note...

'''
midi.config.reset()
s = synth.PatchSynth(6, patch_number=149)
for t in range(20):
	s.note_on(note=36, velocity=1, time = 510 * t)
	s.note_off(note=36, time=510 * t + 500)
    
'''

dpwe, 2025-04-04:
I’m pretty sure it comes from the phase continuity mod in
https://github.com/shorepine/amy/commit/70d8658f0d261925be2829c5618a08a497c15214
I think I can fix it.

bwhitman, 2025-04-05:
Douglas reports this fixed. Thank you DAn!!!!

closed 2025-04-05 by bwhitman