shorepine/tulipcc #581: T-Deck firmware broken in Oct '24

0atman, 2025-04-16:
Something in this (and all subsequent) firmwares broke support for the T-Deck https://github.com/shorepine/tulipcc/releases/tag/v-oct-2024

The last version I could get working on my T-Deck plus was September https://github.com/shorepine/tulipcc/releases/tag/v-sep-2024

## Description of bootloop
- Keyboard flashes
- No startup beep
- Screen initialises with teal background
- resets

## Install Log

```sh
❯ esptool.py write_flash 0x0 tulip-full-TDECK.bin
esptool.py v4.8.1
Found 1 serial ports
Serial port /dev/ttyACM1
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 8c:bf:ea:19:1b:cc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00e5ffff...
Compressed 15073280 bytes to 2357378...
Wrote 15073280 bytes (2357378 compressed) at 0x00000000 in 57.9 seconds (effective 2083.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

0atman, 2025-04-16:
Thanks so much for supporting the t-deck, that's a really exciting portable music sketchpad!

bwhitman, 2025-04-17:
Hm, that is surprising, I definitely tested it back then. Can you link me to the precise T-Deck model you're using to make sure they haven't done a hardware update since the one I test on ?

0atman, 2025-04-20:
Ah! This is the T-Deck plus, purchased last week. Could that be the difference?

bwhitman, 2025-04-20:
It does look like there are some changes to the plus (and maybe later T-Decks) with the display init, and the Plus has no external i2c. I can't tell what it is that's breaking Tulip on boot, it could be something else as well: https://github.com/Xinyuan-LilyGO/T-Deck

I can't guarantee I can help but if you can find a way to see the startup boot log (via USB or the serial header if it's still there) it should give some details to help us.

0atman, 2025-04-20:
It's the least I could do, here's what the log says:

```
Guru Meditation Error: Core / panic'ed (Cache disabled but cached memory region accessed).

MMU entry fault error occurred while accessing the address 0x42aaaaac (invalid mmu entry)

Core 0 register dump:

PC : 0x403797bb PS : 0x00060734 A0 : 0x3fcc10a0 A1 : 0x3fcc0fe0

A2 : 0x3c3bc844 A3 : 0x3fcc10cc A4 : 0x3fcc10c6 A5 : 0x3fcc10c0

A6 : 0x3fcc10d2 A7 : 0x3c3bc844 A8 : 0x8217d816 A9 : 0x00000000

A10 : 0x3c3bc844 A11 : 0x00000001 A12 : 0x80384245 A13 : 0x3fcb0520

A14 : 0x00060023 A15 : 0x00000003 SAR : 0x00000004 EXCCAUSE: 0x00000007

EXCVADDR: 0xaaaaaaa4 LBEG : 0x4200435c LEND : 0x42004365 LCOUNT : 0x00000002

Backtrace: 0x403797b8:0x3fcc0fe0 0x3fcc109d:0x3fcc10c0 |<-CORRUPTED

Core 1 register dump:

PC : 0x4206f7d3 PS : 0x00060e34 A0 : 0x82068d60 A1 : 0x3fccfbf0

A2 : 0x3c3c8e20 A3 : 0x00000001 A4 : 0x3c3cc790 A5 : 0x3fccfbe4

A6 : 0x3c3c8464 A7 : 0x3c3cc790 A8 : 0x00000006 A9 : 0x3c3cbd68

A10 : 0x3c3c8e20 A11 : 0x0000ffff A12 : 0x3c3cc790 A13 : 0x00000000

A14 : 0x00000080 A15 : 0x00010000 SAR : 0x00000019 EXCCAUSE: 0x00000007

EXCVADDR: 0x00000000 LBEG : 0x4206d65a LEND : 0x4206d661 LCOUNT : 0x00000000

Backtrace: 0x4206f7d0:0x3fccfbf0 0x42068d5d:0x3fccfc10 0x42068e19:0x3fccfc50 0x4206b40b:0x3fccfc80 0x42069ba7:0x3fccfcc0 0x42069bf5:0x3fccfd30 0x42069be1:0x3fccfd50 0x42069be1:0x3fccfd70 0x42069c60:0x3fccfd90 0x42069c83:0x3fccfdb0 0x4205c0f3:0x3fccfe00 0x4204717d:0x3fccfe20 0x42014e59:0x3fccfe40 0x42014f2d:0x3fccfe60 0x40378d92:0x3fccfe80 0x4200d9c3:0x3fccff20 0x42014e59:0x3fccff90 0x42014f2d:0x3fccffb0 0x40378d92:0x3fccffd0 0x4200d9c3:0x3fcd0070 0x42014e59:0x3fcd00d0 0x42014f2d:0x3fcd00f0 0x40378d92:0x3fcd0110 0x4200d9c3:0x3fcd01b0 0x42014e59:0x3fcd0210 0x42014e6e:0x3fcd0230 0x420a60b6:0x3fcd0250 0x420a6105:0x3fcd0280 0x420a6322:0x3fcd02b0 0x42014fad:0x3fcd03a0 0x4037901f:0x3fcd03e0 0x4200d9c3:0x3fcd0480 0x42014e59:0x3fcd04e0 0x42014e6e:0x3fcd0500 0x420a60b6:0x3fcd0520 0x420a6105:0x3fcd0550 0x420a6322:0x3fcd0580 0x42014fad:0x3fcd0670 0x4037901f:0x3fcd06b0 0x4200d9c3:0x3fcd0750 0x42014e59:0x3fcd07b0 0x42014e6e:0x3fcd07d0 0x420a60b6:0x3fcd07f0 0x420a6105:0x3fcd0820 0x420a6322:0x3fcd0850 0x42014fad:0x3fcd0940 0x4037901f:0x3fcd0980 0x4200d9c3:0x3fcd0a20 0x42014e59:0x3fcd0a50 0x42014e6e:0x3fcd0a70 0x420394a0:0x3fcd0a90 0x42039841:0x3fcd0b20 0x4200741d:0x3fcd0b50

ELF file SHA256: d4dee207f

Rebooting...

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)

Saved PC:0x40375bf8

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce2820,len:0xf44

load:0x403c8700,len:0x4

load:0x403c8704,len:0xa44

load:0x403cb700,len:0x2b88

entry 0x403c8880
```

dlasky, 2025-04-20:
I saw this thread, flashed the linked build then did an OTA and it seems fine fwiw. I'll try to get a data point from a direct flash as well. (Tdeck plus)

0atman, 2025-04-21:
Huh, I too did an OTA update from a previous build, and that started the bootlooping for me. I don't remember exactly which build I first installed...